Spark-Concepts/xPro-V5 #83: problema cncjs o escala de los archivos

jose1094, 2021-11-15:
tengo un problema con el tamaño de mis archivos tengo una maquina de 1500x1500 pero monto un archivo de 160mm x 70mm y es mas grande que la maquina. espero alguien me pueda ayudar. muchas gracias.

[image]

[image]

[image]

MichaelMartinez, 2021-11-30:
No entiendo el problema. Una vez que encuentre la máquina 0,0, debe configurar el 0,0 de la pieza de trabajo. Si es más pequeño que el área de trabajo, funcionará bien. Le sugiero que busque algunos tutoriales básicos antes de comenzar. ayudarán enormemente.

jose1094, 2022-03-14:
Es qke creo que el tamaño de la maquina en el rgbl está mal yo pongo un
archivo de 120mm*75mm y se sale mas grande que el tamaño de la maquina y la
maquina es de 1500*1500.
Muchas gracias

closed 2022-08-09 by Spark-Concepts