Spark-Concepts/xPro-V5 #84: macros

jose1094, 2021-11-15:
buenas espero que esten bien.
que macros  debo utilizar para una maquina de 1500 x 1500

muchas gracias.

MichaelMartinez, 2021-11-30:
Qué macros necesitas? El tamaño de la máquina se puede configurar antes de ejecutarlos.

jose1094, 2022-03-14:
Hola espero estés bien es qke soy nuevo en esto del cnc y no e podido
funcionar mi maquina tengo una bulkman 150*150 impulsada por correas y no
se como hacer la configuración de tamaño de la maquina cuando pongo un
archivo en mm me ocupa todo el campo de la maquina o se pasa del espacio.

Espero me pueda colaborar.

Spark-Concepts, 2022-08-09:
You will need to change the following settings:
```
$100= (x, step/mm)
$101= (y, step/mm)
$102= (z, step/mm)
```
Type $$ in the command window and make note of the following micro step settings
```
$160 = (x, microsteps)
$161 = (y, microsteps)
$162 = (z, microsteps)
```

... and will depend on the stepper motor your using - for example: typical NEMA 17 motors are rated at 200 steps per revolution and you will need to do some math:
1. Move each axis by rotating the  the applicable stepper motor 1 revolution and measure the displacement in mm
2. Divide 200 by the displacement in mm you determined in step 1
3. Multiply and record the value from step 2 above with the applicable axis microstep value (this will be your $100, $101, and $102 value(s)

for example (x axis):
- NEMA 17 motors are rated at 200 steps per revolution
- ex. $160 = 8
- ex. 20mm per x stepper revolution
- 200-steps/revolution ÷ 20mm/revolution = 10-steps/mm
- 10-steps/mm x 8-microsteps = 80-steps/mm
- Type ```$100=80``` in command window 
- You will do this for each axis

closed 2022-08-09 by Spark-Concepts