Spark-Concepts/xPro-V5 #85: Y2 Limit Switch not registering

ausip, 2021-11-18:
I have an XYYZ machine with 24V NC limit switches on each axis, and limit switches on the both the Y Axis rails.

Y1 limit switch is working perfectly. Y2 is not registering. Below is my initialisation macro for CNC.JS

$Homing/Cycle0=Z
$Homing/Cycle1=X
$Homing/Cycle2=Y
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/DirInvert=Y
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=X
$Homing/Squared=Y
$Homing/Feed=100.000
$Homing/Seek=1000.000
$Homing/Pulloff=5.000
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$X/MaxRate=3000.000
$Y/MaxRate=3000.000
$Z/MaxRate=2500.000
$A/MaxRate=3000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=150.000
$X/MaxTravel=1250.00
$Y/MaxTravel=810.00
$Z/MaxTravel=210.00
$A/MaxTravel=810.00
$X/Current/Run=1.950
$Y/Current/Run=1.950
$Z/Current/Run=1.950
$A/Current/Run=1.950
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8

Mboss32, 2021-11-23:
In the default firmware the Y2 limit is not active, this is to match most kit setups that run a single Y limit with either NO or NC.  

The Y2 can be turned on by uncommenting it in the Machine file and compiling new firmware.   https://github.com/Spark-Concepts/xPro-V5-Firmware

ausip, 2021-11-23:
Thanks for the response. I have had a look through the following file and can't seem to find which commented line relates to the A or Y2 limit switch.

xPro-V5-Firmware/CNCxPROv5_20210708/src/Machines/CNC_xPRO_V5_Machine_Template.h

Is this the correct file?
Please specify which file and which line of code I need to remove the comment on?

Thanks

Mboss32, 2021-11-23:
That's correct,  change the A_Limit_pin declaration to Y2_LIMIT_PIN

ausip, 2021-11-23:
Thanks, I'll give it a go. Slightly off topic.

If I am using a Folinn VFD and wish to use RS485, do I need to make changes to this section when re-compiling? Or can this be taken care of using my initialisation script?

/*********** Spindle Setup******************/
//Default: PWM spindle control 
#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_25
#define SPINDLE_ENABLE_PIN      GPIO_NUM_4
#define DEFAULT_SPINDLE_RPM_MAX     12000 

//Option: RS485 spindle - comment out the PWM spindle block above, uncomment the below block .  
//#define SPINDLE_TYPE            SpindleType::H2A // only one spindle at a time (HuanYang, YL620, or H2A) 
//#define VFD_RS485_TXD_PIN		GPIO_NUM_4
//#define VFD_RS485_RXD_PIN		GPIO_NUM_25
//#define DEFAULT_SPINDLE_RPM_MAX     24000 
/********************************************/

ausip, 2021-11-24:
Hi @Mboss32

I compiled the new firmware, but I had to use PlatformIO, as I was getting compiling errors in Arduino IDE.

The firmware compiled correctly and I have Y2 working perfectly. Now unfortunately when I fire up the spindle using an M3 command the console loops with a ton of messages and I cannot stop the spindle using an M5, I have to his the eStop. The only source change I made was changing the LimitSwitch from A to Y2 & changing the eStop from NO to NC.

Below is my initialization script I am running in CNC.JS and an extract of the console loop after running the M3 command. Using the pre-compiled firmware the spindle works perfectly.

CNC.JS init macro
`$Homing/Cycle0=Z
$Homing/Cycle1=X
$Homing/Cycle2=Y
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/DirInvert=Y
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=X
$Homing/Squared=Y
$Homing/Feed=100.000
$Homing/Seek=1000.000
$Homing/Pulloff=5.000
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$X/MaxRate=3000.000
$Y/MaxRate=3000.000
$Z/MaxRate=2500.000
$A/MaxRate=3000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=150.000
$X/MaxTravel=1250.00
$Y/MaxTravel=810.00
$Z/MaxTravel=210.00
$A/MaxTravel=810.00
$X/Current/Run=1.950
$Y/Current/Run=1.950
$Z/Current/Run=1.950
$A/Current/Run=1.950
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$Spindle/Type=PWM
$GCode/MaxS=24000
$GCode/MinS=8000`

Console log after running M3
`EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff7
Core 1 was running in ISR context:
EPC1    : 0x400912ce  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40092d48
Backtrace: 0x40092d48:0x3ffbe560 0x40091f20:0x3ffbe580 0x40090087:0x3ffbe5a0 0x4018bf61:0x3ffbe5e0 0x4018ca35:0x3ffbe620 0x400dfe6d:0x3ffbe650 0x400dacf2:0x3ffbe670 0x400d938e:0x3ffbe690 0x400d943e:0x3ffbe730 0x400dfbd2:0x3ffbe7f0 0x400dfceb:0x3ffbe920 0x400823c4:0x3ffbe940 0x400824c1:0x3ffbe960 0x40085e01:0x3ffbe980 0x40221cf7:0x3ffbda00 0x400fef4a:0x3ffbda20 0x40091b99:0x3ffbda40 0x400902d9:0x3ffbda60
Core 0 register dump:
PC      : 0x400912d9  PS      : 0x00060034  A0      : 0x80091bd1  A1      : 0x3ffbe300
A2      : 0x3ffbf658  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffc9664
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000001
A14     : 0x00060520  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
[... 942 lines cut ...]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1250.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(-810.000,0.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(-810.000,0.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-210.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Y2 Axis limit switch on pin GPIO(36)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
error:9 (G-code lock)
>`

ausip, 2021-11-25:
Just a follow up. I compile the firmware without error on my Mac using the latest Arduino IDE.

Still having a similar issue with the spindle.
Console error below.

`Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400df294  PS      : 0x00060034  A0      : 0x80081680  A1      : 0x3ffc0670
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xfffffff7  A5      : 0x3ffd1840
A6      : 0x00000020  A7      : 0xffffffe7  A8      : 0x8008157d  A9      : 0x00000001
A10     : 0x00060820  A11     : 0x00000000  A12     : 0x15122500  A13     : 0x00060820
A14     : 0x00000003  A15     : 0x00060823  SAR     : 0x0000001a  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x4008ee9c  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400df294
ELF file SHA256: 0000000000000000
Backtrace: 0x400df294:0x3ffc0670 0x4008167d:0x3ffc0690 0x4008e571:0x3ffc06b0 0x4008ee99:0x3ffd1880 0x4008f33d:0x3ffd18a0 0x4021a5e1:0x3ffd1910 0x40218da1:0x3ffd1930 0x402191e3:0x3ffd1950 0x40217b27:0x3ffd19c0 0x40218542:0x3ffd1a10 0x402172f1:0x3ffd1a70 0x401d0a25:0x3ffd1ab0 0x401d0b0a:0x3ffd1af0 0x401d0b31:0x3ffd1b10 0x400fe55b:0x3ffd1b30 0x400fe677:0x3ffd1b50 0x401f9993:0x3ffd1b70 0x400eef7d:0x3ffd1b90 0x400eacde:0x3ffd1c90 0x400eaf51:0x3ffd1d00 0x400d3fe8:0x3ffd1d20 0x400d2613:0x3ffd1d40 0x400fc22a:0x3ffd1d60 0x40091d02:0x3ffd1d80`

Spark-Concepts, 2021-12-03:
Sorry, I missed the notification on these.  

Was the goal to implement squaring on the Y axis?  If so, this has to be done with the stock firmware using the single Y port - limitation on how the squaring routine is implement for units without Y2 inputs.  

If it not the squaring you need but a true Y2 limit, let me know if you are using a door switch or estop, and if it is normally open, or normally closed, and I will compile and post a .bin for you and post a .bin for you

ausip, 2021-12-03:
The firmware change suggested above to enable the Y2 limit switch for squaring has worked perfectly. After compiling the firmware the auto-squaring works exactly how it should using the following initialization marco in CNC.js

`$Homing/Cycle0=Z
$Homing/Cycle1=X
$Homing/Cycle2=Y
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/DirInvert=Y
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=X
$Homing/Squared=Y
$Homing/Feed=100.000
$Homing/Seek=1000.000
$Homing/Pulloff=5.000
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$X/MaxRate=3000.000
$Y/MaxRate=3000.000
$Z/MaxRate=2500.000
$A/MaxRate=3000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=150.000
$X/MaxTravel=1250.00
$Y/MaxTravel=810.00
$Z/MaxTravel=210.00
$A/MaxTravel=810.00
$X/Current/Run=1.950
$Y/Current/Run=1.950
$Z/Current/Run=1.950
$A/Current/Run=1.950
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$Spindle/Type=PWM
$GCode/MaxS=24000
$GCode/MinS=8000`

The trouble with using this newly compiled firmware is I get the previously posted errors when initiating the spindle. (spindle uses 0-10V and works fine using stock firmware.

The eStop I am using is NC

Spark-Concepts, 2021-12-03:
Thanks for clarification! 

Let's try this firmware (unzip, and then upload the .bin like over the webUI) 

[CNCxPROv5_XYYZ_NC_Y2Lim.zip](https://github.com/Spark-Concepts/xPro-V5/files/7646397/CNCxPROv5_XYYZ_NC_Y2Lim.zip)
e

ausip, 2021-12-03:
[ConsoleOutputErrors.txt](https://github.com/Spark-Concepts/xPro-V5/files/7646750/ConsoleOutputErrors.txt)
I've uploaded the firmware via the WebUI and have attached the console output.

Mboss32, 2021-12-03:
What all do you have plugged into the controller?  I uploaded the same
firmware and can run the pwm spindle without any errors.

Do you have macro buttons?  It looks like those are being triggered/called??

ausip, 2021-12-03:
Currently connected to the controller

XYYZ Stepper Motors
XYYZ Proximity Sensors
Toolhead via relay 
eStop - Door (NC)
24V Power

Probe - Header to connector for XYZ Probe. Nothing plugged into connector
Macro 1 - Header to connector for Tool change Z probe. Nothing plugged into connector

Mboss32, 2021-12-03:
Maybe try pulling out the macro and probe headers at the controller?

ausip, 2021-12-03:
@Mboss32  I'll give it a go but I doubt that's it.

 Why would the Macro triggering only affect it after an M3?
 Running a ? should show that Macro 1 is triggered
If it was something connected to the controller I would have the same issue with the stock firmware

ausip, 2021-12-03:
@Mboss32 , that worked. I disconnected both and it worked. I re-connected probe and it still worked. So you are right, it is the Macro 1. Although it doesn't really make any sense why this would be tripping the controller up.

I will put a meter on the Macro 1 connector I wired and make sure it's not short circuit.

I'd like to still get to the bottom of this, as I want to use Macro 1 for a permanently connected Z-Probe for tool changes.

Mboss32, 2021-12-03:
EMI from the spindle would be my guess - long no terminated leads would
soak that up.

ausip, 2021-12-04:
@Mboss32, you're spot on. It does seem to be EMI. I'm using shielded cable everywhere and have the controller end of the shield grounded.

I'm modifying the shielding now and connecting machine end of the shield to the machine frame, gantry etc. Lets see if that sorts it out.

Once I resolved this Macro 1 interference, I noticed i was getting Hard Limit errors being tripped by the spindle also. Hopefully grounding the machine frame fixes the issues.

Ferdideroo, 2021-12-10:
Hello
i also have the problem, i tried your firmware but then my spindle stops working. It concerns the Huanyang via rs 485 If I enter $ Spindle/Type=HUANYANG, I get an error. I went back to the original firmware there everything works again except making Y axis square with two switches

"Thanks for clarification!

Let's try this firmware (unzip, and then upload the .bin like over the webUI)

CNCxPROv5_XYYZ_NC_Y2Lim.zip"

ausip, 2021-12-17:
@Spark-Concepts  & @Mboss32 . EMI was the culprit for the kernel panic. I grounded the machine end of my sensor cables to the machine frame and it fixed the issue. Originally I only had one end grounded and the machine frame was not.

Y auto-squaring is now working perfectly with the provided firmware. I haven't re-tried the firmware i build, but I am assuming this will work too.

closed 2021-12-21 by Spark-Concepts