Spark-Concepts/xPro-V5 #86: Do these trinamic drivers overheat?

southsideboats, 2021-11-23:
Without fail, every time I run a toolpath that's longer than about 15 minutes,  my Z axis motor starts stalling out on me.... massively. I have the bulkman 3 amp steppers and have the $Z/Current/Run set to 2.7 and $Z/Current/Hold set to 1.0 amps. It's not my wiring, and it's not the motor itself...I have swapped all those things and still get the same issue every single time. Really, really frustrating. I left the microsteps stock at whatever it is and the max speed 3000 mm/min.  100 acceleration... I've done a lot of adjusting the settings and this problem keeps happening..... every single time. Also if I wait until the point where it's missing steps, and then disconnect the coupler from the motor, it STILL stalls, with no load on it.    

Shutting it down for 20 minutes makes the stalling stop happening temporarily.. this is what's led me to believe this is an overheat issue with the tmc5160... 

Oh, I will add that this is repeatable with two separate XPro V5's .

Mboss32, 2021-11-23:
Check your current setting for those motors.  The setting is for RMS current not max.  Discussed here:

https://github.com/Spark-Concepts/xPro-V5/wiki/Changing-settings#xcurrentrun-thru-ccurrentrun-140-thru-145--xyzabc--stepper-run-current-extended-settings

southsideboats, 2021-11-23:
This can be marked as solved.... Totally the issue there.

closed 2021-11-26 by southsideboats