Spark-Concepts/xPro-V5 #87: Won't complete a long toolpath.

southsideboats, 2021-11-23:


southsideboats, 2021-11-24:
This one is really frustrating me.... I had decided that the issue was running the g-code from the SD Card... but today the issue reappeared and this time I was running via USB cable and the file on the computer. It stopped twice in the exact same spot... the countdown in UGS went from 1 hour and 30 minutes remaining to "completed in 10:23". No error messages or anything. Spindle still spinning in the spot it was when it froze.

Mboss32, 2021-11-25:
Can you post your gcode file and how you created it?

southsideboats, 2021-11-25:
> Can you post your gcode file and how you created it?

Here's the file below, created in Aspire. It's not been specific to this one though... I've had different files do the same thing, and one I am pretty sure came from the fusion 360. 

https://drive.google.com/file/d/1rOBB0ufa5UX4Bev9fUTQopN3oxs7M9yj/view?usp=sharing

Mboss32, 2021-11-25:
What post processor did you use?

southsideboats, 2021-11-25:
> 

Just the standard grbl post I thought.

southsideboats, 2021-11-26:
I have duplicated this issue with a second XPro V5 and the toolpath above.

southsideboats, 2021-11-26:
> Can you post your gcode file and how you created it?

Was there an issue with the G-Code I shared? Man, I have been through hell and back getting this whole machine working right.. Finally it's running as it's supposed to but won't finish the job... so it's still just a giant paperweight.

MichaelMartinez, 2021-11-30:
What are your machines GRBL settings? Your motor current settings may be to high, I have seen that a lot in other forums and message boards.  If the motors get too hot, they will shut down.

southsideboats, 2021-11-30:
I think we have confused something here or am I talking to two different
people? Indeed originally my current settings were too high, and bringing
them down to the neighborhood of 1 amp run, .8 amps hold fixed that
particular issue which was loss of steps, major loss of steps. I think they
were set to 2.8 and 1.5 amps respectively originally. That was confusing
for me because of the 3 amp steppers I have and the setting being RMS. I
suggest you make the wiki a bit more clear on that.

southsideboats, 2021-11-30:
> I am jumping in as a bystander to try and help, I don't work for Spark
> Concepts... Just a happy customer. I totally understand having a problem
> and not knowing what to do about it. Going crazy sucks. I will stop
> offering ideas if you want, just ask. For now, I have a few thoughts,
> things to think about and questions...
>
> I looked at the gcode file and it is over 100,000 lines long. Nothing
> stands out to me as being bad in the file. Specifically around the 20,000
> to 40,000 line numbers AKA 29%... They are properly formatted and short
> gcode lines.
>
> Here are some potential ideas as to what may be happening;
>
>    1. Power spike - Does something on the same circuit kick on around
>    this time? Are you turning different things in your space on and off around
>    the time it stops working? Do you have multiple things connected to same
>    circuit? Is the circuit isolated? Is there a possibility of dirty power
>    going to the machine? - A good surge protector will eliminate power spikes.
>    2. Background task when using the computer? Is the computer a laptop
>    or desktop?
>    3. What do run for a spindle? If its a VFD, do you have the spindle
>    properly grounded? Is it on a separate circuit? Is the cable crossing the
>    stepper cables?
>    4. EMI from all the electronics being too close and/or not
>    isolated/grounded properly
>    5. Do you have a rats nest of cables crossing? Are your cables
>    shielded? If not, do you have ferrite on them?
>
> We don't have $250k machines and consequently... our machines are very
> sensitive to electricity and electronic noise. I suspect power is the
> problem here but I am admittedly not an expert.

No, this is all great stuff. I was just wondering if I was talking to one
or two people. I had a bit of a rat's nest of wires, not as bad as the
picture looked, but really that was just because I was switching between
two separate XPro V5's on the fly, trying to eliminate my specific board
being the issue. And since I can duplicate this on both boards, chances of
that being the case is highly unlikely. I am about to hit send with the
spindle turned off and set to no spindle, and see if we can either rule out
or place blame on the RS485 Connection. I will report back. Thank you very
much.

southsideboats, 2021-11-30:
Dammit all to hell. I guess that wasn't it.   Check out the output that I
screenshotted... Straight from Run at 6.41 percent completed, to Idle. I
didn't touch a thing.
[image: image.png]

southsideboats, 2021-11-30:
I saw that also about the SD card formatting, and I did that. In fact, the
SD card in there now is a 4gb microsd card that's older and slower, which I
also read might help. Anyways, I did use that exact software to format it.

Notice also that the Spindle speed stays at 12000... I had disabled it
completely that time with $Spindle/Type=none ... but this gives you an idea
of how the toolpaths are ending up... It just randomly stops, the spindle
keeps right on spinning.

So frustrating man.  CNC is a newer thing to me, but I'm not new to DiY
electronics and the like and I am not a total retard.  I feel a bit like a
beta tester at this point and I've just wasted so much time and gotten
nowhere. I know it's frustrating from the your end as well, that's not lost
on me.. and I do appreciate the help.. Hell, my wife is about to kick my
ass... she's all 'nice CNC, too bad it doesn't work' lol. So many hours
into this and nothing to show for it. I can't help at this point but wonder
where I would be if I had gone with the Mach3 controller instead of the
XPro. That's not personal at all... this is going to get me curses and
frankly I may very well be dead wrong.. I get it, the thing runs on your
test machine so what the hell is wrong with this dude (me)... I wish I
knew. I still can't get past the fact that it does this on two separate
XPro V5's . That would lend you to think I screwed up something on the
hardware side but man, I have been through it all.. and rewired it
several times already. I can jog that sucker all day and it's smooth as you
could want.

I think I'm gonna go to the liquor store now. Lol.

MichaelMartinez, 2021-11-30:
I couldn't see your picture, I think you may be replying from email. There is an issue thread here: https://github.com/Spark-Concepts/xPro-V5/issues/87

How is the machine plugged in? I have mine on two circuits. One for the controller, computer, and power supply for the controller. Another circuit for my VFD/Spindle and water pump. There is nothing else on those circuits. 

I also laid out my wires such that my VFD is about 3 feet away from the computer, xPro and power supply. My wifi was all screwed up on the computer that controls the xPro when the VFD was near it... that is when I discovered EMI problems. I didn't even consider that and it turns out these hobby machines are like EMI sponges while being EMI flame throwers at the same time.

southsideboats, 2021-11-30:
> I couldn't see your picture, I think you may be replying from email. There is an issue thread here: #87
>
> How is the machine plugged in? I have mine on two circuits. One for the controller, computer, [...]

maybe you are onto something here with this. I've got a 3 way splitter coming out of a single outlet. and a 10 foot heavy duty extension cord lol. Water pump I haven't been turning on most of the time.. But I can confirm, even if shamefully, that I've had Xpro, VFD, and Laptop cable all hooked up to that one splitter. Nobody is perfect lol.

Mawg1, 2021-11-30:
>  EMI flame throwers 

the most accurate description of VFDs I have ever seen!

southsideboats, 2021-11-30:
> I couldn't see your picture, I think you may be replying from email. There is an issue thread here: #87
>
> How is the machine plugged in? I have mine on two circuits. One for the controller, computer, [...]

Tried her again.. Still via an extension cord to the power supply  for the XPro. But the VFD was not plugged in and I disabled the spindle in the firmware ($Spindle/Type=none) ...So I'm just moving the thing around in the air now.  Unsurprisingly, she quit on me again. Here's what I'm seeing in the WebUI.. You can see the change from Run to Idle.  

<Run|MPos:214.125,125.615,-27.510|FS:1961,12000|WCO:103.000,103.000,-23.000|SD:4.52,/singlefile.gcode>
<Run|MPos:293.220,125.615,-45.870|FS:3000,12000|Ov:100,100,100|A:S|SD:4.57,/singlefile.gcode>
<Run|MPos:277.240,125.915,-45.870|FS:3000,12000|SD:4.57,/singlefile.gcode>
<Run|MPos:199.500,125.915,-28.530|FS:995,12000|SD:4.63,/singlefile.gcode>
<Run|MPos:144.930,125.915,-44.615|FS:751,12000|SD:4.72,/singlefile.gcode>
<Run|MPos:133.180,127.115,-45.870|FS:3000,12000|SD:5.32,/singlefile.gcode>
<Run|MPos:195.070,129.215,-28.820|FS:1337,12000|SD:6.31,/singlefile.gcode>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000|WCO:103.000,103.000,-23.000>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000|Ov:100,100,100|A:S>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000>
<Idle|MPos:170.470,129.515,-27.965|FS:0,12000>

Mawg1, 2021-11-30:
Maybe try pulling the gcode file from the SD card and then uploading it to NCviewer?  (https://ncviewer.com/)  take a look and see if anything strange is going on there? 

How did you upload the file to the SD card - from the pc or through the WebUI?

southsideboats, 2021-11-30:
> Maybe try pulling the gcode file from the SD card and then uploading it to NCviewer? (https://ncviewer.com/) take a look and see if anything strange is going on there?
> 
> How did you upload the file to the SD card - from the pc or through the WebUI?

Yeah I've been through that already. This is multiple different gcode files now... and the kicker is it also does the same exact thing if I skip the SD card altogether and run the code via USB and either CNCjs or UGS.

MichaelMartinez, 2021-11-30:
Dang. 

What kind of machine is it? The reason I ask is because you have a FS of 3000. My max feedrate is 2000 set in GRBL. I have a Queenbee pro on linear rails with a lead screw. 

Can you please post your GRBL settings as well? type $$ into your command line on the WebUI and copy/pasta that into here.

MichaelMartinez, 2021-12-01:
Did you try lowering the feedrate? Just a thought that maybe your program is outrunning the capability of the steppers torque and/or max rotational speed. There is a lot to digest when it comes to those variables. Ie. Critical speed, coefficient of friction, torque, Coefficient of Collar Friction, etc.

southsideboats, 2021-12-01:
I did just now, to no avail. Man, I went in today with a completely new
toolpath fresh out of Vectric Aspire, standard grbl post. Set the
machine at 2000 max all axes. Got about 8 percent through, stopped, "job
completed in (15) minutes".... It stops where it is in the toolpath and the
spindle keeps spinning (It's disconnected and unplugged but you can see the
"S11000" in the output window.)  ....thing is possessed I swear !

MichaelMartinez, 2021-12-02:
Please post the GRBL setttings you are using...

southsideboats, 2021-12-02:
> Please post the GRBL setttings you are using...

$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=
$Homing/Cycle1=
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=Off
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.020
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=3.000
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=HUANYANG
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2000.000
$Y/MaxRate=2000.000
$Z/MaxRate=2000.000
$A/MaxRate=2000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=200.000
$Y/Acceleration=200.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=1340.000
$Y/MaxTravel=1255.000
$Z/MaxTravel=135.000
$A/MaxTravel=1255.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.000
$Y/Current/Run=1.000
$Z/Current/Run=1.100
$A/Current/Run=1.000
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.800
$Y/Current/Hold=0.800
$Z/Current/Hold=0.900
$A/Current/Hold=0.800
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
Ok

southsideboats, 2021-12-03:
Ran it today with nothing connected... no motors, no spindle, just the emergency stop switch. Same thing happened... it would appear at the very least that this is something not related to my build. Only took me taking the whole thing apart three times to find this out lol.

MichaelMartinez, 2021-12-04:
I am not sure how you got those awesome settings in one shot... all I can manage to do are the GRBL numbers. In any case, we have some differences between our settings:

Of note... step pulse time for me is 4, idle delay is set to 250 

My $X/Current/Run=1.8 and $X/Current/Hold=0.5
 these are the same  on X, Y and Z

[12:44:55] [ $$ ]$0=4 ;Step pulse time, microseconds

[12:44:55] [ $$ ] $1=250 ;Step idle delay, milliseconds

[12:44:55] [ $$ ] $2=3 ;Step pulse invert, mask

[12:44:55] [ $$ ] $3=3 ;Step direction invert, mask

[12:44:55] [ $$ ] $4=0 ;Invert step enable pin, boolean

[12:44:55] [ $$ ] $5=1 ;Invert limit pins, boolean

[12:44:55] [ $$ ] $6=1 ;Invert probe pin, boolean

[12:44:55] [ $$ ] $10=1 ;Status report options, mask

[12:44:55] [ $$ ] $11=0.010 ;Junction deviation, millimeters

[12:44:55] [ $$ ] $12=0.002 ;Arc tolerance, millimeters

[12:44:55] [ $$ ] $13=0 ;Report in inches, boolean

[12:44:55] [ $$ ] $20=0 ;Soft limits enable, boolean

[12:44:55] [ $$ ] $21=0 ;Hard limits enable, boolean

[12:44:55] [ $$ ] $22=1 ;Homing cycle enable, boolean

[12:44:55] [ $$ ] $23=0 ;Homing direction invert, mask

[12:44:55] [ $$ ] $24=100.000 ;Homing locate feed rate, mm/min

[12:44:55] [ $$ ] $25=300.000 ;Homing search seek rate, mm/min

[12:44:55] [ $$ ] $26=250.000 ;Homing switch debounce delay, milliseconds

[12:44:55] [ $$ ] $27=10.000 ;Homing switch pull-off distance, millimeters

[12:44:55] [ $$ ] $30=24000.000 ;Maximum spindle speed, RPM

[12:44:55] [ $$ ] $31=8000.000 ;Minimum spindle speed, RPM

[12:44:55] [ $$ ] $32=0 ;Laser-mode enable, boolean

[12:44:55] [ $$ ] $100=199.000 ;X-axis steps per millimeter

[12:44:55] [ $$ ] $101=200.000 ;Y-axis steps per millimeter

[12:44:55] [ $$ ] $102=200.000 ;Z-axis steps per millimeter

[12:44:55] [ $$ ] $103=200.000 ;undefined

[12:44:55] [ $$ ] $104=100.000 ;undefined

[12:44:55] [ $$ ] $110=2000.000 ;X-axis maximum rate, mm/min

[12:44:55] [ $$ ] $111=2000.000 ;Y-axis maximum rate, mm/min

[12:44:55] [ $$ ] $112=2000.000 ;Z-axis maximum rate, mm/min

[12:44:55] [ $$ ] $113=1000.000 ;undefined

[12:44:55] [ $$ ] $114=1000.000 ;undefined

[12:44:55] [ $$ ] $115=1000.000 ;undefined

[12:44:55] [ $$ ] $120=50.000 ;X-axis acceleration, mm/sec^2

[12:44:55] [ $$ ] $121=50.000 ;Y-axis acceleration, mm/sec^2

[12:44:55] [ $$ ] $122=50.000 ;Z-axis acceleration, mm/sec^2

[12:44:55] [ $$ ] $123=200.000 ;undefined

[12:44:55] [ $$ ] $124=200.000 ;undefined

[12:44:55] [ $$ ] $125=200.000 ;undefined

[12:44:55] [ $$ ] $130=940.000 ;X-axis maximum travel, millimeters

[12:44:55] [ $$ ] $131=1440.000 ;Y-axis maximum travel, millimeters

[12:44:55] [ $$ ] $132=110.000 ;Z-axis maximum travel, millimeters

[12:44:55] [ $$ ] $133=300.000 ;undefined

[12:44:55] [ $$ ] $134=300.000 ;undefined

[12:44:55] [ $$ ] $135=300.000 ;undefined

Mboss32, 2021-12-04:
$s to show all the extended settings,  $$ to show grbl settings

MichaelMartinez, 2021-12-04:
Thanks @Mboss32 !

I just had another thought after finishing a project... Static Electricity. 

My machine shocked the **** out of me when I touched the rail on the y-axis trying to get a look. When I say ****, I mean that in the scared sense. There isn't enough voltage to kill me as the spindle is properly grounded... just that static electricity is obviously building up and I need to chase it down and eliminate it.

southsideboats, 2021-12-06:
> 

Hey, I'm willing to try anything. I will say that cutting this foam that I've been using for test material is very static-y for sure.

southsideboats, 2021-12-07:
> Thanks @Mboss32 !
>
> I just had another thought after finishing a project... Static Electricity.
>
> My machine shocked the **** out of me when I touched the rail on the y-axis trying to get a look. When [...]

The static just doesn't account for the fact that I am not able to complete these paths without the spindle even plugged in.

southsideboats, 2021-12-10:
I'm marking this closed. Replacement board seems to have solved the issue.

luckypants123, 2022-02-02:
Long shot but have a look at [this](https://docs.openbuilds.com/doku.php?id=docs:blackbox:install-windows-selective-suspend)

Tuckvcarve, 2022-03-17:
New to this forum so you can take this with a grain of salt if you wish.
Do you own your copy of Vectric Aspire? The trial version limits the number of useable G code lines. 
If your machine just stops your lucky, I had an older version (trial) that would run just fine for awhile on small jobs but on bigger jobs it would eventually plunge the bit down into my work till it snapped off. 
I ended up buying an copy to use and it never happened again.

luckypants123, 2022-03-17:
Just as an fyi to anyone still struggling... I gave up on the xPro V5 and bought a BlackBox - no more issues.

southsideboats, 2022-03-17:
Yeah definitely not Vectric that was the cause. I gave up also a while back
on that controller. Too much time wasted.

bn1, 2022-10-01:
Hi guys, any news on this? @Spark-Concepts, etc. I've just finished the RatRig's KillerBee with this controller unit and what a heck, I was not able to finish even the wasteboard. Any workarounds? Should I ask someone to return me money for the controller unit and buy another brand? Whats the solution? Will FluidNC solve anything?

PS: I'm running the latest firmware - with the previous one, I was barely able to upload anything to the SD card...

PPS: I can see the very same behavior as mentioned above - it just jump into "idle" state, keeping Spindle (VFD) and cooling running. No error message, nothing. When running the same file again and again, it stops in a different position (everytime after a complete move/command).

Thank you for any advice! It took me ages to get this up and running, well, not so running...

southsideboats, 2022-10-01:
> 

I got an email notification for this question, and I feel your pain, so let me just hit you straight... Get a different controller. Burn this one, put it in the trash, run it over with your car.. doesn't matter. I don't know what the cause or solution is, or why some people don't seem to have issues at all.. I went through THREE of these controllers, hundreds of dollars in wasted material.. and finally just gave up. Do you want to troubleshoot, change firmware a hundred times, deal with people who assume it's you who is doing something wrong?  Or do you want to use the CNC to actually make stuff? 

Just buy something else. I found a shapeoko controller on ebay for a hundred bucks and it hasn't once given me this issue, or any issue for that matter. Same code too. This controller the XPro V5.... is a complete and utter piece of trash.

luckypants123, 2022-10-01:
As Southside said, replace the control. Not worth the frustration.

bn1, 2022-10-01:
Thank you guys for quick responses. I've already spent another few hours tackling things around without any further progress. I'm slowly accepting the reality 🤦‍♀️

luckypants123, 2022-10-01:
I did find attaching ferrite rings to each end of the USB cable made a significant improvement, but not enough that i wanted to keep using it.

bn1, 2022-10-01:
Well, I gave it last try, reorganized cables, gave it new power cable to be as far as possible from the power source, tried it without any other device or tool under electricity but laptop (running on battery) - meaning no vaccum, no air filtration, just nothing else. Running from SD card. No difference...

PS: @luckypants123 thanks for the tip, that's what I'm looking at right now! What's your workflow? CAD/CAM --> usb stick --> hit the play button on the interface?

luckypants123, 2022-10-01:
It’s a nice setup. The interface has a USB port, which you can use, but also wifi, so I just leave the stick in the interface and transfer over wifi.  Then you can use the screen to select files on the stick and run them. Nice as you can leave common files like facing and probing routines on the stick and can run the machine without any pc/laptop/etc…just on its own. 
The interface has all the stuff you’d normally need to setup like probing, jogging, setting zero etc. I’d recommend getting their probe too as it has cycles built in for it. I didn’t and wrote my own cycles for a cheap eBay probe, but if I could do it again I’d get theirs.

281Er, 2022-10-25:
Thank god I ran into this section. I've been chasing ghost for days trying to fix this issue. I pretty much rewired my entire machine but nothing has worked. I have those random program stops. Does any other controller work similar to this where I can connect via wifi to transfer my program to the controller? Thanks in advance

luckypants123, 2022-10-26:
@281Er see my previous. BlackBox + Interface.

bn1, 2022-10-26:
@281Er Well, I have ended up using BlackBox with spare computer attached to it which runs OpenBox Control - then I can connect to it using my phone for jogging as well as sending and running gcode directly from other computer - which is crucial for me. This is IMHO the best setup I can imagine so far for now.

I also bought the Interface, but I'm not using it - there's no possibility of organizing stored files remotely, so then when you are playing around with something, multiple gcode files pile up really quickly, while you are not able to easily delete or organise them from the Interface, so then you need to scroll down the whole list of files, which is really annoying. You are not able to replace file with same name, so every iteration means a file with a new name. Also, the Intefrace is unable to run scripted macros, so you are unable to find hole center, measure, etc., you are able only to find predefined corner of the stock material, nothing more.

Having to say, OB BlackBox + Interface both worked really well since the very first time and I had no issue so far for now. Sadly, I can't say anything similar about SC xPro, I was playing around with it with no better success...

Keegjay, 2023-01-07:
Same problems here with the xpro v5 on ratrig killerbee, has spark concepts just abandoned their customers ? Has anyone had responses from Rat rig ?

southsideboats, 2023-01-07:
> 

Unfortunately this seems to be the case. I hate to say it, but your best bet is to just bite the bullet and save yourself the heartache... get a different controller. Wish I had better news. I have two of these xpros and I wouldn't give them away in good conscience.

281Er, 2023-01-07:
Yes it seems spark concepts has abandoned us. I did however got everything to work after doing all the recommended modifications. Twisted pair cable and signal repeater did the trick for me. Haven't had any issues for a while now.

Keegjay, 2023-01-07:
> Yes it seems spark concepts has abandoned us. I did however got everything to work after doing all the recommended modifications. Twisted pair cable and signal repeater did the trick for me. Haven't had any issues for a while now.

It is a good thing that you got it going though.

Could you give me a reference or a link to the signal repeater you have used ?
Do you use any EMI filter, ferrites ?

What I think is happening is that the xprov5 is getting problems with reading data due to interference because once I get the issue where the machine just stops and says job done, I have to restart the controler or It has a hard time opening/launching and refreshing the files on the sd car while trying to browse from cnc 3d commander or through wi fi webui. It slowly starts showing files but it is slow at getting to show all of them if not reset (I then very frustratingly lose my homing).

281Er, 2023-01-08:
I don't have any emi filters or ferrites. Here's the link for the repeater, just a heads up this one doesn't come with a power cord. https://www.amazon.com/gp/aw/d/B078NTWFRS?psc=1&ref=ppx_pop_mob_b_asin_title

thineight, 2023-01-12:
Hi, about the isolator/repeater, can this be powered by 2 wires 24V DC from the xpro transformer? Or does it require a dedicated power supply?

281Er, 2023-01-12:
I had not thought about but I don't see why I would work if you can adapt it. I believe the repeater can work from 9v to 36v dc.

nguyen4186, 2023-01-19:
Complete my setup about a week ago and experienced the same issue, where the job stop without any error display. This issue been giving me headache, email Spark-Concepts and so far they are eager to help either, ordered a new Blackbox controller and see how it go.