Spark-Concepts/xPro-V5 #91: Spindle speed change during toolpath

ausip, 2021-12-18:
I am having issues with the spindle speed changing during cutting different elements of a toolpath. Spindle speed drops during arcs and ramping. The issue only occurs when using a spindle speed above 10000rpm. It's very noticeable at 24000rpm.

My VFD is connected via 0-10V and I have tested the output and the xPro is tuned to exactly 10v at 24000rpm.

Below are my init macro script etc. for CNC.js & the gcode I am using. gCode is created from Fusion360. I've tried both their included GRBL post processor as well as OpenBuilds post processor.

Here is a video demonstrating the issue.
[https://youtu.be/p8f5Kx_i-Gw](url)

CNC.js Init Script
`$Homing/Cycle0=Z
$Homing/Cycle1=X
$Homing/Cycle2=Y
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/DirInvert=Y
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=X
$Homing/Squared=Y
$Homing/Feed=100.000
$Homing/Seek=1500.000
$Homing/Pulloff=2.500
$X/StepsPerMm=200.002
$Y/StepsPerMm=199.823
$Z/StepsPerMm=200.400
$A/StepsPerMm=199.823
$X/MaxRate=3000.000
$Y/MaxRate=3000.000
$Z/MaxRate=2500.000
$A/MaxRate=3000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=150.000
$X/MaxTravel=1232.50
$Y/MaxTravel=830.00
$Z/MaxTravel=210.00
$A/MaxTravel=810.00
$X/Current/Run=1.950
$Y/Current/Run=1.950
$Z/Current/Run=1.950
$A/Current/Run=1.950
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$Spindle/Type=PWM
$GCode/MaxS=24000
$GCode/MinS=8000`

GCode $$
`client> $$
$0=5 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=2 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=1 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=1500.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=8000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=200.002 (X-axis travel resolution, step/mm)
$101=199.823 (Y-axis travel resolution, step/mm)
$102=200.400 (Z-axis travel resolution, step/mm)
$103=199.823
$104=100.000
$105=100.000
$110=3000.000 (X-axis maximum rate, mm/min)
$111=3000.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=3000.000
$114=1000.000
$115=1000.000
$120=150.000 (X-axis acceleration, mm/sec^2)
$121=150.000 (Y-axis acceleration, mm/sec^2)
$122=150.000 (Z-axis acceleration, mm/sec^2)
$123=150.000
$124=200.000
$125=200.000
$130=1232.500 (X-axis maximum travel, millimeters)
$131=830.000 (Y-axis maximum travel, millimeters)
$132=210.000 (Z-axis maximum travel, millimeters)
$133=810.000
$134=300.000
$135=300.000
ok`
[LeadTableBracketBase.txt](https://github.com/Spark-Concepts/xPro-V5/files/7739084/LeadTableBracketBase.txt)

Spark-Concepts, 2021-12-20:
I sounds like you are getting a little noise through the 0-10V line when under lots of motor loading.  If you put a capacitor on from the 0-10V terminal to the GND terminal on the toolhead connector it should clean up the signal.

ausip, 2021-12-20:
Any particular size capacitor?

Reply to: Spark-Concepts/xPro-V5 ***@***.***>
Date: Monday, 20 December 2021 at 1:05 pm
To: Spark-Concepts/xPro-V5 ***@***.***>
Cc: Paul Mitchener ***@***.***>, Author ***@***.***>

PAUL MITCHENER
Director

Phone
1300 828 747
+61 3 7018 1000
Website
www.ausip.net.au

Mobile
0419 588 320
+61 419 588 320
Email
PO Box 447, Collins St West. VIC 8007

This communication is intended only for the person to whom it is addressed and may contain confidential material. If you received this communication in error, please inform the sender immediately and delete all copies.
​
​Please think of the environment before printing this email.
​

ausip, 2021-12-21:
@Spark-Concepts , thanks for the response. a 22uF capacitor across the input on the VFD fixed it.

I've ordered some ferrites and an AC noise filter to further clean up any other EMI issues.

Thanks heaps

closed 2021-12-21 by Spark-Concepts