Spark-Concepts/xPro-V5 #92: Y2 limit switch not working.

Ferdinandderoo, 2021-12-21:
Hello
i also have the problem, i tried your firmware but then my spindle stops working. It concerns the Huanyang via rs 485 If I enter $ Spindle/Type=HUANYANG, I get an error. I went back to the original firmware there everything works again except making Y axis square with two switches

"Thanks for clarification!

Let's try this firmware (unzip, and then upload the .bin like over the webUI)

CNCxPROv5_XYYZ_NC_Y2Lim.zip"

Ferdinand de Roo

Spark-Concepts, 2021-12-21:
what error do you get with $Spindle/Type = HuanYang?

Ferdinandderoo, 2021-12-22:
I'm very sorry I re update the firmware and everything is working the spindle and the limit switch y2 is working.
Don't no why but i have tried to get the spindle working with a lot of trouble and no success.
I only comment the homing/square=y and homing/cycle2=y
And all is working great didn't need to do everything to get the spindle working.

Spark-Concepts, 2021-12-22:
Great news!

closed 2021-12-22 by Spark-Concepts