Spark-Concepts/xPro-V5 #93: Can you add default firmware for Y2 limit switch?

water-rhatt, 2021-12-23:
Could you add a default firmware for Y2 limit switch? I can compile a custom firmware, but I love that this is as close to plug and play as there is. Updating the git to include a profile for machines with dual Y gantry/auto square/dual Y home switches would be a real help and I can see has come up as an issue several times.

Spark-Concepts, 2021-12-31:
Maybe, we are in the process of converting over to the FluidNC firmware, if that is going to be a long road we will look at pushing the Y2 limit firmware separately.  The big issue is folks using NC switches but only connecting the standard 3 switches will cause some trouble.  

For axis squaring, this can be done with the default firmware using 2 switches on the Y limit port.  Reference info here: https://github.com/Spark-Concepts/xPro-V5/issues/12

water-rhatt, 2021-12-31:
Converting over to FluidNC will make all these machine definitions irrelevant, so I guess it doesn't make sense. I got tripped up on that thinking that maybe my sensor or wiring was faulty. Eventually I realized what was going on. I just thought having a couple of more common pre-compiled machine definitions would be easy enough to create and post and could save on some headaches and support requests. Again, no point if you are switching over to FluidNC.

Thanks.

closed 2021-12-31 by water-rhatt