Spark-Concepts/xPro-V5 #95: Possible to configure as XXYZ?

luckypants123, 2021-12-28:
Is it possible to flip the X and Y axis assignments somehow?

I assume I need to tweak something in the firmware and compile, just looking for pointers on the correct way to do this.

Any help would be appreciated, thank you!

luckypants123, 2021-12-29:
I also found the below in GCode.cpp:
```
                     case 'X':
                        if (n_axis > X_AXIS) {
                            axis_word_bit               = GCodeWord::X;
                            gc_block.values.xyz[X_AXIS] = value;
                            axis_words |= bit(X_AXIS);

                        } else {
                            FAIL(Error::GcodeUnsupportedCommand);
                        }
                        break;
                    case 'Y':
                        if (n_axis > Y_AXIS) {
                            axis_word_bit               = GCodeWord::Y;
                            gc_block.values.xyz[Y_AXIS] = value;
                            axis_words |= bit(Y_AXIS);
                        } else {
                            FAIL(Error::GcodeUnsupportedCommand);
                        }
                        break;
```

Would swapping the axis assignments here have the desired effect as well?

Spark-Concepts, 2021-12-31:
you should do this in the Machine configuration file (first post).  Just change Y2 to X2, then you will have an X Y X2 Z setup so most of the silkscreen will still be correct

luckypants123, 2021-12-31:
Awesome, thank you!

closed 2021-12-31 by luckypants123

lagnat, 2022-04-02:
@luckypants123 Where you successful swapping X and Y?  Did you have to rebuild the firmware?

luckypants123, 2022-04-02:
Yes, made the changes above and then compiled it following the directions in the wiki. 
wasn’t too difficult and gave me an XXYZ machine.

lagnat, 2022-04-02:
What version of Esp32 did you install?  bdring/Grbl_Esp32 docs say 1.0.3 and nothing higher but it looks like they're at 2.0.2 so wondering if those are stale docs.

luckypants123, 2022-04-02:
I believe I used the source available from the Spark Concepts GitHub. It’s definitely behind in version, but I didn’t want to experiment with using the code from somewhere else.

lagnat, 2022-04-02:
The Spark Concepts README says to follow the instructions at the bdring/Grbl_Esp32 repo, however it also says to follow the instructions at https://github.com/espressif/arduino-esp32.  Can you check which version of Esp32  you used?

https://github.com/Spark-Concepts/xPro-V5-Firmware#compiling-firmware

luckypants123, 2022-04-02:
About 2 weeks after this the controller died, so that marked the end of my xProV5 adventure. 

If I remember right though, you need to follow the instructions at both those places, as they each cover a different part of the process. However instead of using the source code from bdring/, I used it from the Spark Concepts GitHub, as that version of the Arduino sketch has all the pin assignments correct for the unit. 

To be honest I don’t remember exactly, but I do remember it was not clear.