Spark-Concepts/xPro-V5 #97: How to enable M4 command

toonaoki, 2022-01-01:
I'm using the XPro-V5 in PWM mode. I'm trying to switch the spindle with 0-10V analog signal and the laser with PWM.

Firmware used: CNC_xPRO_V5_XYAZ_PWM_NO.bin

When I run the M4 command
"error: 20 (Unsupported command)"
”[MSG: M4 requires laser mode or a reversable spindle]”
Is displayed.

I couldn't use the M4 command even if I enabled the $ 32 laser mode.

$Spindle/Type=LASER
$GCode/LaserMode=1
I also tried these commands,
”[MSG: Warning: LASER_OUTPUT_PIN not defined”
Is displayed and PWM becomes the maximum output.

How do I enable the M4 command?
Is it possible that this firmware is not supported?

toonaoki, 2022-01-01:
I solved it myself.
By rewriting to the latest firmware, M4 is enabled in Laser mode.

What is the difference between normal firmware and BM3D Laser firmware?

Spark-Concepts, 2022-08-09:
Done...

closed 2022-08-09 by Spark-Concepts