Spark-Concepts/xPro-V5 #98: Long programs stop partway through

luckypants123, 2022-01-01:
Just got my controller and I can't seem to get longer programs to complete.
If the program is short, everything works great... however if the program is longer, it just stops partway as if feed hold was pressed.
I'm pretty sure the controller is crashing somewhat as immediately after this happens the Web UI is unavailable.

I've tried running via SD Card and CNCjs, same result.

I thought it could be electrical interference, so I didn't run any ancillary equipment (vacuum), but same result.

It looks to me like [#87](https://github.com/Spark-Concepts/xPro-V5/issues/87)

My NC program:
[1001.txt](https://github.com/Spark-Concepts/xPro-V5/files/7798150/1001.txt)

My Settings:
```$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.000 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$100=1600.000 (X-axis travel resolution, step/mm)
$101=1600.000 (Y-axis travel resolution, step/mm)
$102=1600.000 (Z-axis travel resolution, step/mm)
$103=1600.000
$104=1600.000
$105=1600.000
$110=2500.000 (X-axis maximum rate, mm/min)
$111=2500.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=2500.000
$114=2500.000
$115=2500.000
$120=600.000 (X-axis acceleration, mm/sec^2)
$121=600.000 (Y-axis acceleration, mm/sec^2)
$122=600.000 (Z-axis acceleration, mm/sec^2)
$123=600.000
$124=600.000
$125=600.000
$130=265.000 (X-axis maximum travel, millimeters)
$131=535.000 (Y-axis maximum travel, millimeters)
$132=110.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
$140=1.000
$141=1.000
$142=1.800
$143=1.800
$144=0.250
$145=0.250
$150=0.050
$151=0.050
$152=0.050
$153=0.500
$154=0.125
$155=0.125
$160=64
$161=64
$162=64
$163=64
$164=64
$165=64
$170=16
$171=16
$172=16
$173=16
$174=16
$175=16
```

Any advise would be most welcome!

Ian.

CraftyMyner, 2022-01-02:
I am also having similar issues. Both UGS and OpenBuilds Control lock up after a while of being connected, no more serial coms. I have to disconnect and reconnect via the button in the interface. It appears the controller didn't lose position so it must just be communication.

Running macOS, OpenBuilds Control, and UGS.

luckypants123, 2022-01-03:
Also not sure if its related or not (or something I don't quite yet understand about the controller), but occasionally on startup the Z axis will not move... other axes will.

I command homing, the readout shows the Z axis values moving positive, but the axis is stationary. After a few seconds an alarm is posted due to not finding the Z limit switch.

The only way I can fix it is to reboot the ESP32.

allistar, 2022-01-07:
I also have the "Z axis not responding on initial connection" issue and the only fix I've found is to disconnect and then reconnect.

For the issue with long programs crashing: I had a similar issue and it was caused by the USB connection dropping. The XPro doesn't like that. I replaced the cheap USB cable with a VR headset one and I also stopped running it alongside the 230V power cable. That made the issue go away.

CraftyMyner, 2022-01-07:
I added some ferrites on the USB cable and that seemed to do the trick for me. While attempting to fix the issue I tried getting FluidNC working for the xPro V5 but with only mild success. Would be good to get an official release from @Spark-Concepts going forward.

luckypants123, 2022-01-07:
> I also have the "Z axis not responding on initial connection" issue and the only fix I've found is to disconnect and then reconnect.
>
> For the issue with long programs crashing: I had a similar issue [...]

I've had this issue both running over USB and also running of the internal SD Card. I can't remember if the USB cable was plugged in when I was running off the SD Card though. Are you saying that even when running off the SD Card, just the act of having the USB connected would cause it?

I did take the precautionary measure of isolating the USB cable from any high power sources as well.

luckypants123, 2022-01-07:
> I added some ferrites on the USB cable and that seemed to do the trick for me. While attempting to fix the issue I tried getting FluidNC working for the xPro V5 but with only mild success. Would be good to get an official release from @Spark-Concepts going forward.

I'll order some ferrites... hopefully that helps.
I saw on another closed issue they are looking at FluidNC but are waiting for it to mature a bit. I'm looking forward to that as apparently it has good support for squaring the gantry with 2 limit switches.

allistar, 2022-01-07:
I've never used the SD card before so I'm not sure on that. When I have the shop vac plugged in close to the computer it can cause the USB to momentarily disconnect. It would be good if the controller handled this scenario. Looking at the kernel panic it causes this could be at a lower level. I have also had the shop vac trip the limit switches which can be annoying because CNCjs doesn't let you restart the gcode from an arbitrary line like LinuxCNC does.

luckypants123, 2022-01-07:
That’s good info, thank you. I’ll do some experiments over the weekend with different USB cables, routing, ferrite, etc and see if I can isolate the issue.

CraftyMyner, 2022-01-07:
I tried to use the SD card for a 625 hole operation and it worked but about every 5 holes you would hear a slight studder like it was starved for the next move. Using the supplied SD card, haven't looked to see what speed it is but I don't think that's the issue.

luckypants123, 2022-01-07:
> 

I haven't seen that so far - are you interpolating the bore helically with short linear moves maybe? Even so if it completes 1 hole without being data starved the time to rapid to the next hole would be plenty to refill the buffer. Granted my experience is with industrial controllers and I'm just starting with grbl, but I'd imagine the theory is the same.

CraftyMyner, 2022-01-07:
> I haven't seen that so far - are you interpolating the bore helically with short linear moves maybe? Even so if it completes 1 hole without being data starved the time to rapid to the next hole [...]

All holes are done with a G3, but fusion seems to have made them using about 20-30 commands per hole. The holes are very close together. Once I fixed the USB issue and used it instead of SD I had no issues with stuttering.

southsideboats, 2022-01-07:
> Just got my controller and I can't seem to get longer programs to complete. If the program is short, everything works great... however if the program is longer, it just stops partway as if feed hold was pressed. I'm pretty sure the controller is crashing somewhat as immediately after this happens the Web UI is unavailable.
> 
> I've tried running via SD Card and CNCjs, same result.
> 
> I thought it could be electrical interference, so I didn't run any ancillary equipment (vacuum), but same result.
> 
> It looks to me like [#87](https://github.com/Spark-Concepts/xPro-V5/issues/87)
> 
> My NC program: [1001.txt](https://github.com/Spark-Concepts/xPro-V5/files/7798150/1001.txt)
> 
> My Settings:
> 
> ```
> $1=255 (Step idle delay, milliseconds)
> $2=0 (Step pulse invert, mask)
> $3=0 (Step direction invert, mask)
> $4=0 (Invert step enable pin, boolean)
> $5=0 (Invert limit pins, boolean)
> $6=0 (Invert probe pin, boolean)
> $10=1 (Status report options, mask)
> $11=0.010 (Junction deviation, millimeters)
> $12=0.002 (Arc tolerance, millimeters)
> $13=0 (Report in inches, boolean)
> $20=1 (Soft limits enable, boolean)
> $21=0 (Hard limits enable, boolean)
> $22=1 (Homing cycle enable, boolean)
> $23=3 (Homing direction invert, mask)
> $24=200.000 (Homing locate feed rate, mm/min)
> $25=2000.000 (Homing search seek rate, mm/min)
> $26=250.000 (Homing switch debounce delay, milliseconds)
> $27=2.000 (Homing switch pull-off distance, millimeters)
> $30=24000.000 (Maximum spindle speed, RPM)
> $31=0.000 (Minimum spindle speed, RPM)
> $32=0 (Laser-mode enable, boolean)
> $N1=
> $N0=
> $33=5000.000
> $34=0.000
> $35=0.000
> $36=100.000
> $100=1600.000 (X-axis travel resolution, step/mm)
> $101=1600.000 (Y-axis travel resolution, step/mm)
> $102=1600.000 (Z-axis travel resolution, step/mm)
> $103=1600.000
> $104=1600.000
> $105=1600.000
> $110=2500.000 (X-axis maximum rate, mm/min)
> $111=2500.000 (Y-axis maximum rate, mm/min)
> $112=2500.000 (Z-axis maximum rate, mm/min)
> $113=2500.000
> $114=2500.000
> $115=2500.000
> $120=600.000 (X-axis acceleration, mm/sec^2)
> $121=600.000 (Y-axis acceleration, mm/sec^2)
> $122=600.000 (Z-axis acceleration, mm/sec^2)
> $123=600.000
> $124=600.000
> $125=600.000
> $130=265.000 (X-axis maximum travel, millimeters)
> $131=535.000 (Y-axis maximum travel, millimeters)
> $132=110.000 (Z-axis maximum travel, millimeters)
> $133=300.000
> $134=300.000
> $135=300.000
> $140=1.000
> $141=1.000
> $142=1.800
> $143=1.800
> $144=0.250
> $145=0.250
> $150=0.050
> $151=0.050
> $152=0.050
> $153=0.500
> $154=0.125
> $155=0.125
> $160=64
> $161=64
> $162=64
> $163=64
> $164=64
> $165=64
> $170=16
> $171=16
> $172=16
> $173=16
> $174=16
> $175=16
> ```
> 
> Any advise would be most welcome!
> 
> Ian.

I'm the one who was having this issue on the other thread....After 3 Xpro v5's all doing the same thing... stopping in the middle of long programs. 

I spent a month trying everything imaginable and it just kept happening. For the time being, my solution was to use a shapeoko controller instead. I'm just trying to save you the hassle here... until something changes, this is an issue if you're trying to run long toolpaths. And a critical one.

luckypants123, 2022-01-07:
Ah well that’s not promising. 
I thought from that thread it looked like a new board resolved it. :/

I’m going to do some experiments this weekend. Hopefully as mentioned above it is some type of sneaky EMI issue.

southsideboats, 2022-01-08:
> Ah well that’s not promising. I thought from that thread it looked like a new board resolved it. :/
> 
> I’m going to do some experiments this weekend. Hopefully as mentioned above it is some type of sneaky EMI issue.

I thought the new board solved it, but it reared its ugly head again shortly thereafter.  Mine was even not finishing long toolpaths after I disconnected all the motors and spindle and ran it on the bare controller basically. I will follow along here.... You seem to be a bit more knowledgeable than me and maybe you can be the one that helps figure this out.

luckypants123, 2022-01-09:
Ok, well definitely some progress.

As @CraftyMyner suggested I went after the USB cable, and that seems like it is/was the problem.

I did 2 things:

1. Replaced the USB cable with a shorter one. Originally I was using whatever I had on hand (maybe a ~10ft cable?). I replaced with a 5ft that would just reach. Maybe the additional length was inducting noise like an antenna?
2. I added a decent clip-on size ferrite bead to both sides of the cable, right butt up to the connectors. The ones I used were 6mm ID, 29mm long. Similar to [these on Amazon](https://www.amazon.ca/iSKUKA-Pack-Ferrite-Beads-Clip/dp/B06XVR81CR/ref=sr_1_1_sspa?crid=P3DKYMK3PQCB&keywords=ferrite%2Bcore%2Bsnap%2Bon%2B6mm&qid=1641753693&sprefix=ferrite%2Bcore%2Bsnap%2Bon%2B6mm%2Caps%2C81&sr=8-1-spons&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyMUJZRThGMzBaSVNLJmVuY3J5cHRlZElkPUEwMjkxMzAwR1k5NVVZSTNVMUtZJmVuY3J5cHRlZEFkSWQ9QTA1NDk1MDFKTTgzUFhJMVZPT0omd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl&th=1). I picked mine up local but basically the same thing.

Previously my longest runtimes were around 15min, less if the shop vac was running.

As a test I ran a long program (56min, 127k lines, helical boring, 24k RPM, 2500ipm). Completed successfully.
Then reran with shop vac on, and again completed successfully.

Not sure if it's 100% resolved yet, but I think @Spark-Concepts should definitely note in the setup wiki that ferrite beads on the USB cable are recommended.

I'll leave this issue open for another week or so just to verify and then hopefully can be closed.

CraftyMyner, 2022-01-09:
> Ok, well definitely some progress.
> 
> As @CraftyMyner suggested I went after the USB cable, and that seems like it is/was the problem.
> 
> I did 2 things:
> 
> 1. Replaced the USB cable with a shorter one. Originally I was using whatever I had on hand (maybe a ~10ft cable?). I replaced with a 5ft that would just reach. Maybe the additional length was inducting noise like an antenna?
> 2. I added a decent clip-on size ferrite bead to both sides of the cable, right butt up to the connectors. The ones I used were 6mm ID, 29mm long. Similar to [these on Amazon](https://www.amazon.ca/iSKUKA-Pack-Ferrite-Beads-Clip/dp/B06XVR81CR/ref=sr_1_1_sspa?crid=P3DKYMK3PQCB&keywords=ferrite%2Bcore%2Bsnap%2Bon%2B6mm&qid=1641753693&sprefix=ferrite%2Bcore%2Bsnap%2Bon%2B6mm%2Caps%2C81&sr=8-1-spons&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyMUJZRThGMzBaSVNLJmVuY3J5cHRlZElkPUEwMjkxMzAwR1k5NVVZSTNVMUtZJmVuY3J5cHRlZEFkSWQ9QTA1NDk1MDFKTTgzUFhJMVZPT0omd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl&th=1). I picked mine up local but basically the same thing.
> 
> Previously my longest runtimes were around 15min, less if the shop vac was running.
> 
> As a test I ran a long program (56min, 127k lines, helical boring, 24k RPM, 2500ipm). Completed successfully. Then reran with shop vac on, and again completed successfully.
> 
> Not sure if it's 100% resolved yet, but I think @Spark-Concepts should definitely note in the setup wiki that ferrite beads on the USB cable are recommended.
> 
> I'll leave this issue open for another week or so just to verify and then hopefully can be closed.
> 
> If it comes back I have one other idea to try. I wonder if the injection molded case of the xPro might not be shielding the components enough, leading to some noise there. If the issue comes back, I'll bend up some extruded metal to make a faraday cage and place the xPro in there and see if that help. It will kill the wifi, but I prefer using gSender anyway.

Good to hear,

I had given up on my machine as the longest job I was able to run was less than a few minutes. With the release of FluidNC, I thought I would take another stab at it. When testing the spindle at various RPM's I found the issue would show up, specifically at around 10k with an almost 100% fail rate with. I then threw a couple of ferrites on and didn't have the issue show up after 50 tests. I then ran a 2hr hole drilling program with an unrelated gcode failure (too low precision for arc commands). I think it's fixed for me but I am worried it might crop up if all the stars align. (Steppers, Spindle, Air Compressor, etc.)

Might give the inside of the case a spray of conductive paint or copper tape.

allistar, 2022-01-09:
Are your spindle cable and stepper motor cables all fully shielded? It's odd the issue seems related to spindle speed. My VFD is about 300mm away from the XPro and I use fully shielded spindle and stepper wires. I've run gcode files with > 1 million lines taking 12 hours. The only issue I have is every now and then the machine will pause for a few seconds, it's as if the controller is starved of commands but eventually it catches up.

CraftyMyner, 2022-01-09:
> Are your spindle cable and stepper motor cables all fully shielded? It's odd the issue seems related to spindle speed. My VFD is about 300mm away from the XPro and I use fully shielded spindle and [...]

@allistar 

The cables are all fully shielded for the steppers, VFD, and limit switches. All shields are bonded to earth (I tried floating and bonded to the negative but no difference) The VDF is about 500mm away from the controller, the vfd and stepper cables run in the same cable chain with 2 waterlines in between. the USB cable takes a completely different route on the other side of the table to the laptop. Tried with laptop plugged (grounded) and unplugged.

luckypants123, 2022-01-09:
Not sure if that’s what you mean by the VFD cabling but I’ve run my spindle power cable and waterlines in its own cable chain on the opposite side of the gantry to keep it as far as possible from the low voltage stuff. Maybe that’s the cause?

allistar, 2022-01-09:
@CraftyMyner Your setup sounds similar to mine. Here's a picture of my Y axis drag chain. The water is running inside the Y axis chain but outside the X axis chain (there wasn't enough room). Other than that, all cables are running together in one drag chain. I've made sure the spindle cable shielding is wired to the earth in the VFD.
[image]

CraftyMyner, 2022-01-09:
Most spindles have a variable frequency drive (VFD) that chops up the AC to vary the speed of the spindle. The cheap Chinese ones are usually very electrically noisy and cause havoc with electronics. Shield cables are a must for the cable from the VFD to the spindle, also recommended the supply for the VFD takes another route from the low voltage cables.

CraftyMyner, 2022-01-09:
@allistar Yes, quite similar. I managed to get my waterline into my cable chains so I put it between the spindle cable and the stepper cables. I have the limit switches running with the stepper motor cables. I had to add an extra external pull-down resistor for the limit switches as I was getting false positives. I switched my Y axis to inductive type limit switches and don't need the pull-downs anymore on them.

allistar, 2022-01-09:
@CraftyMyner That's interesting, my electrically noisy shopvac is tripping one of my limit switches. My next step it to diagnose which one. I've assumed the static in the vac is causing a small current to form in the limit switch wiring which is causing the false positive. Can you explain (or send a pic or link) of how you used a pull-down resister to fix this?

CraftyMyner, 2022-01-09:
@allistar You can see the standard dry contact switch (Z) beside one of the inductive switches (Y2). I have a 10k pulldown resistor from signal to ground, the switch is connected to positive and signal on the controller and, Com and NC (normally closed) on the switch. The inductive switches did not seem to have a problem with false positives and I omitted the pulldown. If you are using NPN type inductive switches then you should use a pullup (resistor from signal to positive) instead. Inputs configured as NC.

[image]

allistar, 2022-01-10:
Hi @CraftyMyner , thanks very much for the information. I've got some testing to do to see if this fixes the issue.

luckypants123, 2022-01-26:
Well I've run a bunch and that specific issue seems to be gone for good after liberal use of ferrite beads on the USB cable. I'm calling this one solved.

closed 2022-01-26 by luckypants123