Spark-Concepts/xPro-V5 #99: TROUBLE CONNECTING HUANYANG VFD WITH XPRO V5 CONTROLLER USING RS485 COMUNICATION

salcantarat, 2022-01-04:
I buy a Bulkman CNC QueenBee  with the CNC Xpro V5 controller and HUANYANG VFD to control a 2.2KW spindle water cooled.
Here is my Issue: When I start my spindle with M3 command, the spindle reach the velocity (I checked using a tachometer), but then the CNCjs send a error message and I'm not able to change the speed and continue to send g commands either.

> M3 S7000
error:9 (G-code lock)
> $X
[MSG:Caution: Unlocked]
ok
> M3 S7000
[MSG:Spindle RS485 Unresponsive 7]
[MSG:Critical Spindle RS485 did not reach speed 7000. Reported speed is -1 rpm.]
ok
ALARM:10
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
error:9 (G-code lock)
[MSG:VFD: VFD settings read: RPM Range(6000, 24000)]]

How can I fix it?

*** In the following lines I explain my set up:
1.	Wiring VFD with Xpro V5
2.	HUANYANG VFD register settings
3.	Xpro V5 settings
4.	CNCJS g code sender, $$ , $S

**1 Wiring VFD with Xpro V5**

VFD R -> 220V neutron
VFD S -> none
VFD T -> 220V line
VFD U -> 2.2 KW spindle
VFD V -> 2.2 KW spindle
VFD W -> 2.2 KW spindle
*** EARTH WIRING NOT CONNECTED
*** JUMPER VFD HUANYANG IN VI POSITION
VFD RS485+ -> A Xpro V5 (with ferrite chokes)
VFD RS485- -> B Xpro V5 (with ferrite chokes)

[image]

**2 HUANYANG VFD register settings**

PD000   0
PD003   400
PD004   400
PD005   400
PD011   100
PD014   5
PD015   5
PD023   1
PD141   220
PD142   10
PD143   4
PD144   3000

PD001  2
PD002  2
PD163  1
PD164  1
PD165  3

**3 XPRO V5 SETTINGS**

Update Firmware with CNC_xPRO_V5_XYYZ_NC.bin following the instructions listed here: https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading
Define the spindle with: $Spindle/type=HuanYang
Switch to RS485

**4 CNCJS g code sender, $$ , $S**

CNCjs 1.9.23 [Grbl]
Conectado al puerto COM3 con un baud rate de 115200
cro button only permitted in idle]
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20210816]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,750.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,800.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,800.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-100.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Auto-RTS in use]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:None]
[MSG:VFD: VFD settings read: RPM Range(6000, 24000)]]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]

client> $$

[MSG:'$H'|'$X' to unlock]
$0=4 (Step pulse time, microseconds)
$1=250 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=1500.000 (X-axis maximum rate, mm/min)
$111=1500.000 (Y-axis maximum rate, mm/min)
$112=1500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=30.000 (X-axis acceleration, mm/sec^2)
$121=30.000 (Y-axis acceleration, mm/sec^2)
$122=30.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=750.000 (X-axis maximum travel, millimeters)
$131=800.000 (Y-axis maximum travel, millimeters)
$132=100.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok

> $S

$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=4
$Stepper/IdleTime=250
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.500
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=HUANYANG
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=1500.000
$Y/MaxRate=1500.000
$Z/MaxRate=1500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=30.000
$Y/Acceleration=30.000
$Z/Acceleration=30.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=750.000
$Y/MaxTravel=800.000
$Z/MaxTravel=100.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.500
$Y/Current/Hold=0.500
$Z/Current/Hold=0.500
$A/Current/Hold=0.500
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
Ok

TTALON541, 2022-01-13:
Still nothing huh, i bought the same setup, Bulkma queen bee, xpro, etc, I'm 110v, the same problem, hopefully, it gets addresses

southsideboats, 2022-01-13:
> Still nothing huh, i bought the same setup, Bulkma queen bee, xpro, etc, I'm 110v, the same problem, hopefully, it gets addresses

I would forego the rs485 altogether if I were you and just go the pwm route instead. Save yourself the headache.

spydercnc, 2022-01-16:
Use the 485 firmware I used NO with a NO switch. A NC switch must be in place to use the NC firmware. 
I battled for 3 days, changed the firmware and also changed the wire between the Xpro and the VFD and it all started working. The LED under the Xpro RS485 must flash green/red. 
Switch the VFD on first and then the Xpro.

ColPaulR, 2022-01-22:
I had some problems with getting RS-485 to work as well. I finally have it going. The solution to my problems was to tie the 220V AC ground to the 24V power supply ground and to connect a digital ground (I used the toolhead gnd) to the Huanyang ACM. I verified that removing either of these connections and turning on the VFD creates enough EMI and ground float to cause the RS-485 to quit working. Not sure it will solve your -1 RPM problem but thought I'd give you something to try.

deverelo, 2022-01-26:
I had that exact error of the RS485 Unresponsive 7 and Spindle RS485 did not reach speed.
At that time I didn't have the shield of the spindle cable grounded. I attached the braid to the vfd ground, keeping this pigtail very short (30mm?) at the VFD side. The mains ground and spindle cable ground are also attached to this same terminal.
At the spindle side the connector housing is not attached to ground as it should be. So I exposed the shield and made it have contact with the outer spindle housing (I have 2.2kW air cooled model). Later I will get a small and large metal hose clamp to make a proper connection. Anyway just improving the cable shield fixed this - it is really important with VFDs.

ghalib1399, 2022-02-02:
> I had some problems with getting RS-485 to work as well. I finally have it going. The solution to my problems was to tie the 220V AC ground to the 24V power supply ground and to connect a digital [...]

Dear can you show us image for your setup?

id192313, 2022-06-02:
Hi, able to resolve the issue?

Randall-Dev, 2022-06-13:
whats the pros and cons on using pwm or rs-485

Jennywhite2, 2022-07-16:
> I buy a Bulkman CNC QueenBee with the CNC Xpro V5 controller and HUANYANG VFD to control a 2.2KW spindle water cooled. Here is my Issue: When I start my spindle with M3 command, the spindle reach [...]

> I buy a Bulkman CNC QueenBee with the CNC Xpro V5 controller and HUANYANG VFD to control a 2.2KW spindle water cooled. Here is my Issue: When I start my spindle with M3 command, the spindle reach [...]

Hello, can someone share wiring diagram vfd with power supply,because i dont no how to put ac cable ,..need to put on vfd or power supply

Spark-Concepts, 2022-08-09:
Please see the following [https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453](https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453)

closed 2022-08-09 by Spark-Concepts

id192313, 2022-08-09:
edited 
I have the same issues and implemented all the fixes that have suggested before. Some hit and missed. In the end the only thing that make it works is the resistor on the RS485 (both side) and using a Belkin surge protector. The moment I use a different extension without any grounding it failed. Hope this help.
[image]