Spark-Concepts/xPro-V5 #101: Always having to press the rest button to get started.

spydercnc, 2022-01-15:
Finally got my xPro V5 working with a HY VFD, using  CNC_xPRO_V5_XYYZ_485_NO.bin. on a Lillerbee machine.
When rest all works well, movement, etc is as it should be, connected via USB-C. The spindle turns on/off with the software as expected and machine limits are recognized. 
I have the VFD wired separately that it does not run aimlessly when working on the workpiece or changing bits. 

However, both when I switch the machine on or switch the VFD on later, I have to press the reset button on the xPro to connect to the killer bee or run the spindle. I understood that the VFD needs to be powered on first when initially setting everything up, but pressing the reset button and then rehoming the machine every time I need to change something, seems an unnecessary step.

spydercnc, 2022-01-16:
I was wondering if the Macro connections can be enabled to reset with a push-button switch?

sflom, 2022-01-17:
I would second the idea of a macro to reset the push-button switch.  I have to also press the reset button to get my machine to react like I would like it to often.

Spark-Concepts, 2022-01-20:
Using RS-485 there is a handshake with the VFD and controller, if the VFD is off on startup the handshake will be missed and not allow the spindle to function properly.  There is also a handshake between the drivers and the ESP32 module, if USB is applied before 24V is applied, the handshake will be missed and the ESP32 needs to be restarted. 

Is your xPRO constantly plugged into USB, or the port left open?  In the sender programs, if you select the "Enable Hardware Flow Control" option, the xPRO will auto reset when the port is opened.

spydercnc, 2022-03-12:
Thank you that did work.

closed 2022-08-09 by Spark-Concepts