Spark-Concepts/xPro-V5 #102: Spindle error on my Queen Bee with CNC xPro V5 RS485 control

fredqbee, 2022-01-15:
In my startup message using CNCjs I get:
[MSG: Initializing RS485 VFD spindle]
[MSG: Undefined VFD_RS485_TXD_Pin]
[MSG: Undefined VFD_RS485_RXD_Pin]
[MSG: Undefined VFD_RS 485_RTS_Pin]
[MSG: RS485 VFD Spindle Error]
Are these missing settings in the HuanYang VFD or something else.
Can anybody help?

kongpop113, 2022-01-16:
Here is how I set it up.

1. Follow the wiring instruction on this [youtube link](https://www.youtube.com/watch?v=I2wec_L6j14). 
2. Program VFD to establish RS485 communication [reference](https://buildyourcnc.com/FAQ/13383#:~:text=You%20will%20need%20to%20connect%20the%20VFD%20to,with%20mach3%20called%20the%20Huanyang%20VFD%20Mach3%20PlugIn.)
   PD001 - value 2 - Enables with communication aspect of the VFD
   PD002 - value 2 - Enables frequency control from the communcations port
   PD163 - value 1 - VFD address
   PD164 - value 1 - Baud rate of 9600 (speed of the communication)
   PD165 - value 3 - 8N1 RTU - 8 bits (that's the size of the word that gets communicated), N (No parity or no error checking) and 1 (1 stop bit)
   
Be aware of noise coupling from the spindle to the limit switches. However, connecting the limit switch shields to ground inputs of the V5 controller gets rid of this issue.

fredqbee, 2022-01-18:
Sorry it took so long to respond.I doubled checked the settings on the VFD and they were correct.Frustrated, I did a factory reset of the VFD and then did a bench test of the spindle.It hummed but didn't turn. I checked the wiring and was correct but 1 wire was loose.I tightened the wire and did the bench test. It ran fine.I reset the VFD setting, rechecked the wires to the xpro. All goodI checked the PWM/RS485 switch and flipped it back and forth.There was not much resistance between the two, more of a slide.I started backup and got the same error message.Defined the spindle again trying both HaunYang andH2A.Nothing. Checked the indicator light for the PWM/RS485 and it was very light red almost like a reflection of the door stop button.Flipped the switch to PWM And no difference. Flipped it back and pulled out the green connector and the green light came on just before it was completely out.I pushed it back in and the green light when out. Double checked the wire and they were fine.I could be wrong, but it seems that the PWM/RS485 Switch could be bad.Do you have any ideas?

 
 

Here is how I set it up.

Spark-Concepts, 2022-01-20:
When you have the switch set to RS485, what color LEDs do you see underneath the switch?

fredqbee, 2022-01-20:
I n my previous response, I mention that the LED under the PWM/RS485 switch was a very light red. Almost like the E-stop LED was reflecting through. I hit the E-stop to shut that LED off and there is no light under the PWM/RS485 switch. Also, if I flip the switch to the PWM side I still have no light. The only time I get a light is when I pull out the connector about halfway then the light turns green.

Cyrill87, 2022-02-04:
I have the same setup and do have the same problem. For me the old firmware (containing 485 in name) is working

fredqbee, 2022-02-04:
That firmware is loaded On My Pro V5. I was able to view this when I was testing out CNC Commander. I am not able to view this on CNCjs. CNCjs just connects. It doesn't tell me which firmware it's using, nor do I know how to change it. Would this explain why I don't have a lit led under the PMW/485RS switch.

sdom001, 2022-02-21:
I am having the same issue. I am getting this
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_TXD_PIN]
[MSG:Undefined VFD_RS485_RXD_PIN]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:RS485 VFD spindle errors]
I do have a green light under the Spindle spot on the side of the xpro v5.
[image]
Has anyone figured out what the issue is? I have checked all wiring and all seems fine. What guage wire should be used for the RS485 connection. I am using a small section left over from the limit switches, should it be thicker, does it matter?

sdom001, 2022-02-23:
ok update. I emailed Mike at Spark Concepts and he had me Run the $i command. It sent back this.
[VER:1.3a.20201022:]
[OPT:MPHBSW#]
[MSG:Using machine:CNC_xPRO_V5_XYYZ]
[MSG:Mode=AP:SSDI=CNC_xPRO_V5:IP=192.168.0.1:MAC=94-3C-C6-9D-74-55]
[MSG:No BT]
I sent it to him and his response was my Xpro v5 firmware was out of date and needed to be updated to use the RS485 with spindles. and to go here https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading.
which then led me here https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware   to get the update
and here https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading#ota-over-the-air-firmware-updates for the update instructions. I will post a follow up if it resolves the issue.

fredqbee, 2022-02-23:
Sorry to leave GitHub community in the dark. I contacted Spark Concepts directly and the got me up and running.I have been trying to upload our conversation so everybody can see. But it crashes or says it's too big. I will keep trying and then close this issue as solved

fredqbee, 2022-02-23:
> ok update. I emailed
> Spark ***@***.******@***.***, Jan 27 at 8:14 PMHi Fred,
> Sorry to hear there is trouble, our Github guy is out sick at the moment so we are a bit slow on the responses there - sorry!
> Could you send command $i to the controller and send me the response? Could you send command $s to the controller and send me the response? What VFD are you trying to connect to? could you send a picture of your setup?
>
>
>
> A user has submitted the contact form on your store.
>
> Here are their details:
>
> | Email Address: | ***@***.*** |
> | Phone Number: | 13304170795 |
> | Greetings, I am fredqbee on the GitHub site.
> I have not seen a response from my last post, so I thought I would contact you direct. I have no light under the RS485 switch. I am assuming I have a faulty unit. Everything else works. Is there a way to check the signal from the xPro V5 to the VFD? Can I get a replacement? Where do I go from here? |
>
> Spark Concepts
> https://www.spark-concepts.com/
> Spark Concepts is powered by BigCommerce. Launch your own store for free with BigCommerce.
>
> |
> |
> |
> |  |  |
>
>  |
>
>  |
> |
> |  |
> Ecommerce for a New Era | BigCommerce
>
> Learn how BigCommerce can fuel your business with all the capabilities of enterprise ecommerce—without the cost ...
>  |
>
>  |
>
>  |

fredqbee, 2022-02-23:
> hi MikePlease check your spam for my response.It probably ends up in spam because of attached filesThanks Fred
>
>
>
>
> Hi Fred,
> Sorry for the delay,  found this in spam for some reason.
> Quick question,  if you set the switch to RS 485 side,  unplug the 485 cable going to the VFD completely,  then set:
> $Spindle/Type = HuanYang
> Press the reset button on the side of the control and send me a picture of the output message that pops up in cncjs.  Please also note the color of the LED under the 485 switch with the cable unplugged
> Thanks, Mike
>
>
> Hi Mike,Sorry I am not up on the new way to share photos and screens. I can only attach files. I will give you a little background on my machine. I purchased a QueenBee Pro from Bulkman with your xPro V5 and the 2.2Kw water cooled spindle and the HaunYang VFD. Everything is moving, homes and spindle bench test works fine. I can't get the spindle to turn on with the xPro and Cncjs. Also, the LED under the PMW/RS485 is not lit in either mode, the switch is very loose and if I pull the connector out about halfway the LED does turn green. Checked the VFD settings, Wire connections, changed wire, and spread out all the components.

Hi Fred, 
I see part of the problem, you are using the PWM only firmware - you will need to push an update, please follow these instructions : Checking_firmware_and_upgrading · Spark-Concepts/xPro-V5 Wiki (github.com)
make sure to match the ending NO or NC to the type of limit switch you have when selecting a .bin to download and install.

fredqbee, 2022-02-23:
> Hi Mike,I am a little confused. When I bought the QueenBee, Bulkman said that the xPro V5 had all the firmware already installed on the SD card. I started using cncjs in the beginning, but axis didn't move correctly, the spindle didn't work, and it was a little confusing. After some research I switched to cnc3d commander. I was able to get my axis working and homed. I also was able to see what was on the (pics included) SD card. Unfortunately, I still couldn't run my spindle and cnc3d commander doesn't communicate with fusion 360. I switched back. Luckily, everything I did in cnc3d commander seamed to transfer over. I saw the firmware on the SD card but cncjs doesn't differentiate between firmware. It just loads. If I load another firmware on the SD card, will I be able to pick the correct firmware or will it just load. Why can't I pick the correct one now? Will I be able to upgrade to the WebUI? I am assuming I need to load the CNC_xPRO_V5_XYYZ_485_NO. I don't see any instructions on how to load this to the xPro V5

Hi Fred, 
The xPRO comes loaded with the stock firmware, which can run the spindle using a PWM or 0-10V signal.  To run the Spindle with the RS485 control new firmware needs to be loaded.  
For now ditch CNCcommander - you can come back to that when the firmware is loaded. 
turn on the controller, then open your computers wifi settings. You will see a CNC_xPRO_V5 access point available.  Connect to it, and the embedded Web interface on the controller will open.  
From there, follow these instructions to upload the .bin that you downloaded.  Instructions: Checking_firmware_and_upgrading · Spark-Concepts/xPro-V5 Wiki (github.com)

fredqbee, 2022-02-23:
Best I could. Hope you can follow

sdom001, 2022-02-23:
Updating the firmware to the xpro v5 worked. I used the .NO file. spindle now works

fredqbee, 2022-02-23:
I used CNC_xPro_V5_XYYZ_No.Bin

sdom001, 2022-02-24:
I did as well thank you. The only issue I have now is the spindle spins counter clockwise when it should be turning clockwise.

Silvio Domingues

sdom001, 2022-02-24:
> I think there is a command that inverts the spindle. I may be wrong.When you do a command of $S you get a list of settings.Mine says: $Spindle/Enable/Invert=off, yours may say "on"Also, if I remember correctly there was a setting in the VFD that determined which direction the spindle turnedOtherwise you may have to swap wiresBut again I may be wrong

thanks for the reply. I did a little more research and it is simply swapping any 2 wires from the vfd to the spindle.(u,v,w)