Spark-Concepts/xPro-V5 #103: Z limit switch problem

IlValeSocial, 2022-01-16:
Hi,

I have a similar issue, if I try to press limit switch of Z axis doesn't work, If I try to connect it to xpro board, led don't turn on

If I try to test pressing X limit switch and send "?" command I receive message: 
<Idle|MPos:135.955,-12.500,24.410|F5:0,0|Pn:XZ|WCO:0.000,0.000,0.000>

IF i try with Z limit switch I display this:
<Idle|MPos:135.955,-12.500,24.410|F5:0,0|WCO:0.000,0.000,0.000>

I hope is PIN problem, How can I solve?

Thanks a lot for any help

_Originally posted by @IlValeSocial in https://github.com/Spark-Concepts/xPro-V5/issues/20#issuecomment-1013753425_

IlValeSocial, 2022-01-25:
Hi, there is a solution to my issue?

Thanks a lot

Spark-Concepts, 2022-08-09:
Are you still having issues with your Z limit switch? If so, please re-post and I will be happy to assist :-)

closed 2022-08-09 by Spark-Concepts