Spark-Concepts/xPro-V5 #104: Relay problem

matus1504, 2022-01-18:
Hello.
I have problem with my relay.
I wanted it to turn on the vacuum cleaner when spindle works.
[image]
When the internal jumper i on the "mist signal" relay works on m7 command.
But when I change jumper on "spindle enable signal" and turn on spindle, relay will not work.

Spark-Concepts, 2022-01-20:
Have you defined a spindle type with setting: 

$Spindle/Type = 

Can you share your $s settings?

matus1504, 2022-01-20:
spindle type is Huanyang

$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=4
$Stepper/IdleTime=250
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=On
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=
$Homing/Squared=
$Homing/Feed=100.000
$Homing/Seek=1500.000
$Homing/Debounce=250.000<Idle|MPos:-238.322,23.835,3.812|FS:0,0>
$Homing/Pulloff=3.000
$GCode/MaxS=24000.000
$GCode/MinS=8000.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=HUANYANG
$X/StepsPerMm=400.000
$Y/StepsPerMm=400.000
$Z/StepsPerMm=400.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2000.000
$Y/MaxRate=2000.000
$Z/MaxRate=2000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=300.000
$Y/Acceleration=300.000
$Z/Acceleration=300.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=1305.000
$Y/MaxTravel=1285.000
$Z/MaxTravel=125.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=2.500
$Y/Current/Run=2.500
$Z/Current/Run=2.500
$A/Current/Run=2.500
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.500
$Y/Current/Hold=0.500
$Z/Current/Hold=0.500
$A/Current/Hold=0.500
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok
>

Spark-Concepts, 2022-02-06:
I see your using the RS485 to control your spindle. Unfortunately the "spindle enable signal" is consumed by the RS485 signal (see below)

A work-around as you mentioned would be to move the internal jumper to the "mist signal" - then add an M7 command at the beginning of your g-code to activate the relay for your vacuum.

[image]

closed 2022-08-09 by Spark-Concepts