Spark-Concepts/xPro-V5 #107: Laser mode no pwm signal

Nastyfast44, 2022-01-26:
I am trying to activate the laser using m3 commands. I am able to activate it with $spindle/type=pwm.  I send command m3s24000 and the laser turns on at what I expect is full power, as my max spindle speed is 24000.   The issue occurs when I change to $spindle/type=laser with $32 1. It’s states that laser_output_pin not defined. I send m3s24000 and I do not detect any voltage from pwm pin. Laser will not fire. 

I’m using pwm firmware with a neje a40640 laser module. I am also using ugs to send commands.

Spark-Concepts, 2022-02-06:
You will need to use [$pindle/type=PWM](https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#spindletypepwm); on the xProV5's default firmware there is not a physical GPIO assigned to the "laser" $spindle/type

***Note: the $spindle/type sets which GPIO is currently active as well as if the signal is discrete(ON/OFF), analog(PWM), or RS485***

Regardless ***PWM*** and ***Laser*** Spindle Types are functionally the exact same.

Nastyfast44, 2022-02-07:
I got it to work. Thank you. I realized the newest firmware worked for both pwm and rs485. Switched and everything seems to be working as it should now. Thank you

closed 2022-02-07 by Nastyfast44