Spark-Concepts/xPro-V5 #108: Laser and Rs485 switch

matus1504, 2022-01-26:
How can I choose between spindle and laser without touching the Rs485/Pwm switch?
"Changing between the different control options requires the appropriate firmware to be installed"
What do you mean about it?
I can't find a solution.

Nastyfast44, 2022-01-28:
[image]
Choose the file that relates to your emergency stop switch. The lever on your xpro will swap between the two (pwm& rs485). Don’t use anything on the old firmware page.  Don’t use the laser firmware either unless u have a bulkman laser. 

Link to firmware in case pic doesn’t go through. 

https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware

Nastyfast44, 2022-01-28:
To my knowledge which this fixed my issue 5 min ago. This is a cross between the two and works for both

closed 2022-02-06 by Spark-Concepts