Spark-Concepts/xPro-V5 #109: I have problem with HuanYang!!!

ghalib1399, 2022-02-03:
Hi ALL
when I do $spindle/type=huanyang
I receive:
[MSG:Initializing RS485 VFD spindle]
[MSG:Auto-RTS in use]
E (634349) uart: UART driver already installed
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:None]
ok

after that I do 
M3 S8000
I get 
[MSG:Spindle RS485 Unresponsive 6]
[MSG:Critical Spindle RS485 Unresponsive]
ALARM:10
[MSG:Critical Spindle RS485 did not reach speed 8000. Reported speed is 0 rpm.]
ok
ALARM:10

OR

[MSG:Spindle RS485 Unresponsive 7]
[MSG:Critical Spindle RS485 did not reach speed 8000. Reported speed is -1 rpm.]
ok
ALARM:10

then i have to  do unlock and stop the router from the vfd!!!
-----

I flow [https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#toolheadvfd-rs485](url)
but it not solve my problem

-----
> $$
$0=4 (Step pulse time, microseconds)
$1=250 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=4 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000<Idle|MPos:0.000,0.000,0.000|FS:0,0> (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=8000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=400.000 (X-axis travel resolution, step/mm)
$101=400.000 (Y-axis travel resolution, step/mm)
$102=400.000 (Z-axis travel resolution, step/mm)
$103=400.000
$104=100.000
$105=100.000
$110=4000.000 (X-axis maximum rate, mm/min)
$111=4000.000 (Y-axis maximum rate, mm/min)
$112=4000.000 (Z-axis maximum rate, mm/min)
$113=4000.000
$114=1000.000
$115=1000.000
$120=300.000 (X-axis acceleration, mm/sec^2)
$121=300.000 (Y-axis acceleration, mm/sec^2)
$122=300.000 (Z-axis acceleration, mm/sec^2)
$123=300.000
$124=200.000
$125=200.000
$130=1292.000 (X-axis maximum travel, millimeters)
$131=1205.000 (Y-axis maximum travel, millimeters)
$132=300.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok

Cyrill87, 2022-02-04:
I do have the same Problem with the new Firmware and RS485

Nic-marangon, 2022-02-12:
Me too...how did you resolve?

ColPaulR, 2022-02-12:
> I do have the same Problem with the new Firmware and RS485

Maybe I'm reading into the above, but does the problem not exist with prior firmware? If so, which firmware works? I have the same issue and have switched to 0-10V plus relay control. I'd like to go back to RS485 if we can make it stable.

deverelo, 2022-02-16:
Use shielded 4 core cable for your spindle. Connect the earth and shield at BOTH ends. This is specific to vfds compared to other shielding

Spark-Concepts, 2022-08-09:
please check out the following: [https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453](https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453)

closed 2022-08-09 by Spark-Concepts