Spark-Concepts/xPro-V5 #110: Please advice - how to control brushless spindle motor (wiring)

wildfang, 2022-02-06:
Hello there,
I'm trying to make this brushless spindle motor work with automatic control:

[image]

https://www.amazon.de/KSTE-WS55-220-Brushless-Spindel-BLDC-Motortreiber-Controller/dp/B082563ZJD

It runs fine when controlled with the pot and switch, but I need to connect it to the controller so I can control it by GRBL.

So far I put the selector to EN/PWM and connected the GND and 10V to the matching terminals on the speed controller.
I also connected the EN on the toolhead terminal to the EN on the speed controller.

However, that is where I am now stuck. I don't know how to get it to run.

I tried SpindleType::DAC  as well as SpindleType::_10V.

But nothing. I can hear the relay clicking when sending M3 but the spindle does not move.

For the manual control, the pot is connected to:
GND (black), 10v (red), SV (yellow)

The switch is connected to:
GND (black), F/R (blue)

Any help on how to get it working? That would be great, thanks!

Kind regards

Spark-Concepts, 2022-02-06:
You will first need to configure the spindle as: $Spindle/Type=PWM

The pwm signal is activated by an M3 or M4 gcode statement. The value of the pwm signal is determined by the speed portion of the M3/M4 command – ex. M3 S500 will create a half max pwm signal output or 2.5V on the PWM terminal and 5V on the 0-10V on the terminal (assuming the default max spindle speed of 1,000rpm)... 

***The max spindle speed setting is defined by setting $30 (typically $30-1000)***

See below for wiring (the switch controls the direction of the spindle - per the motor driver's datasheet):

[image]

wildfang, 2022-02-07:
Hi,

thank you very much for your reply. I wired it up via the diagram but I still don't get any output on the 10v terminal.
I can get 5V on the PWM terminal with S3 1000 but the 10v terminal measures at 0.075V. 

Here is what I see when printing settings:

> 
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=PWM

$30 is set to 1000

Do I need to do something further to enable the 10V terminal? I have not tried adjusting the internal pot because I think it can't be that far off from the factory, no?

Spark-Concepts, 2022-02-08:
If you are measuring 0-5V DC on the `PWM` output, you should also measure 0-10V DC on the `0-10V` output. Unless wiring was connected such that the 0-10V was short circuit to ground causing damage. If this was the case the 0-10V op-amp is likely bad and would need to be replaced.

***But before going there***; try manipulating the internal pot as it may be measuring a dead spot. If this doesn't fix your issue please contact us via https://www.spark-concepts.com/contact-us-1/

wildfang, 2022-02-08:
Hi,
thanks for your reply. I don't measure any current on the 0-10V.

However, as I was a little impatient, I just plopped a PWM to 0-10V converter board into my controller and everything works very well.

Is the 0-10v OP-amp some part I could change myself at a later stage, if I happen to need it?

Spark-Concepts, 2022-08-09:
You can - however if you live in the US we can replace it for you free of charge - use the Contact Us link on our Spark-Concepts.com website

closed 2022-08-09 by Spark-Concepts