Spark-Concepts/xPro-V5 #112: Mist allows vacuum turn on during power up or reset

ColPaulR, 2022-02-07:
I have my dust collector controlled by the mist signal through an SSR. If I leave the dust collector switch in the on position, the Xpro V5 will turn on the dust collector until the end of its boot sequence. Same thing happens when I connect a USB. I'm assuming a soft reset on USB connect. I would STRONGLY suggest moving the initialization for this earlier in the startup sequence. Looking through the source code, it appears to me that the pin direction and initial value does not get set until near the end of the sequence. void loop->run_once()->reset_variables()->coolant_init(), which contains pinMode() and coolant_stop().

Maybe I'm reading the code wrong. If not, I suggest both the pinMode() and coolant_stop() be called as soon as possible. Seems like void setup()->grbl_init() would enable the GPIO to be set to a safe configuration right after i2s_out_init().

Spark-Concepts, 2022-02-07:
ColPaulR, 

So in looking at the schematic, yes it appears GPIO (21) if tristate, would cause the MIST output FET to source ground (turn ON). As you can see below there there is an additional nFET (2N7002) which is level shifting the esp32's 3.3V logic to 5V applied to the gate of the larger nFET (Q2B).  This is because a 3.3V gate voltage is insufficient to allow Q2B to fully conduct.

We will take a look at the startup sequence to minimize the initialization period to not have GPIO 21 in tristate throughout the boot sequence.

Alternately if your Spindle Type is set to PWM you can use the EN output. The spindle enable "EN" is activated when the spindle is turned ON - Please note the output of EN is 3.3V (most SSR's control voltages are 3-32VDC).  

[image]

ColPaulR, 2022-02-07:
Thanks for the rapid reply. I'll take a look at using the EN signal vs. mist as I'm using 10V plus relay (couldn't get RS-485 to reliably work). I added an "M7" plus a 2 second dwell to the post processor before M3 to allow the dust collector to turn on and get past the inrush before the spindle turns on and sinks current. The dust collector is 110V on a separate circuit but it is a 2HP and dims the lights a little when spooling up. Spindle/VFD is much better behaved in its init. 

BTW, I'm in Washington Township OH. I noticed that Spark Concepts was Waynesville so we are neighbors!

closed 2022-08-09 by Spark-Concepts