Spark-Concepts/xPro-V5 #114: Spindle problem.

nonosapiens, 2022-02-17:
Hello,

I have a problem with VFD / PWN / Spindle speed. 
When I send M3 the engine runs at 24,000 normally - keep sae speed
But when I send a gcode to Cnc, Spindle speed changes with big variations
When I measure the voltage at the Cnc pro V5 output, the voltage moves ( 9v... 8v.. 7V... )  during the cut
Its note the Gcode, each line have same Sxxxx

I use new firmware  CNCxPROv5_XYYZ_NO
Where can it come from ?

Thanks alot

nonosapiens, 2022-02-17:
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/DirInvert=X
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Feed=100.000
$Homing/Seek=1500.000
$Homing/Pulloff=5.000
$X/StepsPerMm=26.84
$Y/StepsPerMm=26.78
$Z/StepsPerMm=199.100
$A/StepsPerMm=26.678
$X/MaxRate=9000.000
$Y/MaxRate=9000.000
$Z/MaxRate=800.000
$A/MaxRate=9000.000
$X/Acceleration=150
$Y/Acceleration=150
$Z/Acceleration=50
$A/Acceleration=150
$X/MaxTravel=1252.00
$Y/MaxTravel=2484.00
$Z/MaxTravel=80.00
$A/MaxTravel=2484.00
$X/Current/Run=1.9
$Y/Current/Run=1.9
$Z/Current/Run=1.9
$A/Current/Run=1.9
$X/Current/Hold=1.1
$Y/Current/Hold=1.1
$Z/Current/Hold=1.1
$A/Current/Hold=1.1
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$Spindle/Type=PWM

closed 2022-08-09 by Spark-Concepts

bn1, 2022-08-16:
Hi @Ugilten , so how did you connected it? I have the same spindle and I'm about to connect it today. Just realized that I need that 10-26 V from somewhere, so I'm thinking about connecting it to the mist port, while changing the pin for the output power to 24 V. Am I thinking right about it? Should I connect and use both gnd slots in those both ports? Thank you for the inspiration! :)

Ugilten, 2022-08-16:
Hi @bn1 ,
I connected the tool GND to PSU GND and gave the spindle 24V from the PSU as well and that fixed it.

bn1, 2022-08-16:
Hi @Ugilten did you connected it directly to PSU, meaning there are 24 volts all the time? Not through the x5 pro and waiting for any signal? Huh, maybe an easier and more robust solution 😄

Ugilten, 2022-08-16:
Hi @bn1. Yes, the spindle is constantly on. I didn’t think about that before now 😬
Luckily the spindle has the ‘on’ button… but I keep forgetting it…

bn1, 2022-08-16:
@Ugilten Ah, cool, that works for me. I'm switching off power for my whole shop when I'm out, so this is no issue for me. Thank you for the resolution! 😉

Ugilten, 2022-08-16:
No problem @bn1 . Happy cutting 👍

bn1, 2022-08-19:
@Ugilten well, I have connected to mist port to be able to switch on/off spindle programatically. After few uses I have figured out that it is more practical to let in on all the time. I'm still able to switch the spindle off with the button on the spindle itself. So I ended up with your solution 😄 cheers 👌