Spark-Concepts/xPro-V5 #116: Spindle start but doesn't stop after firmware update

JosDrolkoppers, 2022-02-26:
Hi All,

I connected a NC emergency stop and had to update the firmware. Everything works, but I can only start the spindle (via RS458) in Candle, but I can't stop it. 
$Spindle/type=HUANYANG.

Everything worked fine before the update.

Thanks for your thoughts.
BickyBurger

JosDrolkoppers, 2022-05-27:
Thanks for the advice. I tried the twisted pair, but I got the same result.
I noticed you updated the documentention and added a resistor. I'll try that.

KInd regards,
Maarten

id192313, 2022-06-02:
Hi, managed to get it solved?

Randall-Dev, 2022-06-13:
How did you do it?  and what country are you in?  I'm in Europe using 230v the emi is driving me nuts..

id192313, 2022-06-14:
Hi Randall, I started with not able to control the spindle fully. Kept giving unresponsive error after it start running. Than I followed all the fixes I found on GitHub and Reddit. Install new firmware (NC), separate spindle wire away from the rest, ground the spindle wire, put resistor on both of the rs485 end. Still doesn’t work. 
Then I remove the the spindle ground on the VFD and move it to the body ground. Suddenly it works. Everything works. I did some test by moving the spindle cable next to the controller. Didn’t work. I thought I found the culprit but I was wrong cause I put everything back to the cable chain and even remove the ground. It still works. Only thing that did t work is the resistor.  Remove it and the this stop working.
Between these changes I found lose wire on those green connector and reinstall the firmware and reconfigure it. Now the only diff from the beginning is the resistor. That must definitely be part of it. I’m on 240v in Malaysia. 

Hope this help.

closed 2022-08-09 by Spark-Concepts