Spark-Concepts/xPro-V5 #122: Controller has stopped responding, console claims no firmware

bandidobrent, 2022-03-17:
I swapped out my Endurance laser for my SainSmart (done this many times before) but now when I try to connect I get nothing. Tried connecting to the controller, only response I can get through any program is no firmware is present. No way to flash firmware, is my board fried?

Spark-Concepts, 2022-03-18:
**1. What senders (programs) are you using to connect to the xPro?**
**2. Do you see the xPro-V5's COM port present under device manager when you plug the controller in via USB?**
**3. Are you able to access the xPro-V5 WebUI via Wi-Fi?** 

If you're controller is unresponsive via Wi-Fi and USB - is it possible that power (either 24V or USB) was connected while connecting your laser to the controller? If so, it's possible something may have accidentally shorted in the process.
 
That said, if you have a multimeter check a few things before we run down a rabbit hole...

1. Disconnect the USB cable and power only using your 24V power supply
5. Remove the lid by removing the four lid screws and expose the top PCB
6. Turn your 24V power supply ON 
7. Measure for 5 Volts as shown below (also check the remaining voltages as shown on the right)
8. Remove 24V or turn OFF 24V power supply
9. Plug in USB cable, and power via a PC or USB power brick 
10. Re-Measure for 5V and 3.3V (24V should be 0-ish)
11. Please let me know your test results (actual voltages measured)

| Voltage  | 24V PSU Only | USB Only |
| ------------- | ------------- | ------------- |
| 3.3V  | (value?)  | (value?) |
| 5V  | (value?)  | (value?) |
| 24V  | (value?)  | (value?) |

[image]

Also, if you have a USB multimeter (this step is not necessary - but may aid in troubleshooting), please check the Current draw with the USB connected (should be drawing 150-300mA @ ~5V)

[image]

bandidobrent, 2022-03-18:
>    2. Remove the lid by removing the four lid screws and expose the top
>    3. Turn your 24V power supply ON
>    4. Measure for 5 Volts as shown below (also check the remaining
>    5. Remove 24V or turn OFF 24V power supply
>    6. Plug in USB cable, and power via a PC or USB power brick
>    7. Re-Measure for 5V and 3.3V (24V should be 0-ish)
>    8. Please let me know your test results (actual voltages measured)

Always good to be an electrician by trade when it comes to troubleshooting
these things, already had those numbers ready to go for ya.

PSU: 24v-23.9, 3.3-3.3, 5v-4.9
USB: 24v-0, 3.3-3.3, 5v-5
I cannot connect via WebUI. Tried via 192.168.0.1, asks for credentials.
Used the default login, no dice.
I usually just run LaserBurn. last night I ran through the list of senders
(LaserBurn, Candle, OpenBuilds, gSender, LaserGRBL) trying to get some kind
of usable data

Spark-Concepts, 2022-03-18:
Ok, so power looks good - you can try resetting the device to it's factory default firmware by using the [Flash Download Tool](https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware/Old%20Firmware/flash_download_tool) 

It's important that you only use the firmware binary found in the same folder as the flash tool (be sure to read the Read-Me file)

bandidobrent, 2022-03-18:
The flashing tool has been running for over an hour now, pretty sure the controller isn't taking the firmware.... got anything else to try?

Spark-Concepts, 2022-03-18:
Be sure you hold down the button labeled "PGM" prior to clicking start on the flash tool...

If it doesn't automatically sync and begin uploading.. i.e. you see a message ```sync... ... ...```, you may need to nudge it into programming mode...

To manually enter programming mode do the following:
1. Hold down the recessed button labeled "PGM" next to the USB-C connector
2. While holding down the "PGM", toggle the "Reset" button next to the power switch 
3. Keep holding down the the PGM until it begins uploading (you may need to toggle reset a few times - all the while holding the PGM button)

bandidobrent, 2022-03-18:
I tried that last night, no luck. Tried this morning, no luck.

Spark-Concepts, 2022-03-18:
Copy - please contact us directly at https://www.spark-concepts.com/contact-us-1/ for repair/replacment

closed 2022-03-18 by Spark-Concepts