Spark-Concepts/xPro-V5 #126: 0-5v output ttl not above 2.7v

TheFoxInBowen, 2022-03-30:
Since new i could not get a ssr relay to stay in, so i ran my router via the relay output.
I have now added a laser and want to use the ttl output to drive it.
In lightburn i turn on constant power mode, 100% power, and he ttl output should be 5v approx and i get 2.6v.
Anyone had that issue?
Thanks
Trevor Fox

Spark-Concepts, 2022-05-16:
When you say ttl output are you referring to the terminal labeled "PWM"?

- performing the same test (at 100% power) 
  1. measure the "0-10V"
      - do you measure ~5V or ~10V?  
  2. power everything down and try toggling the "PWM - RS485" switch a few times to ensure it's indexed correctly and making good electrical contact (the PWM ttl signal routes through that switch).

Spark-Concepts, 2022-08-09:
Are you still having this issue?

closed 2022-08-09 by Spark-Concepts