Spark-Concepts/xPro-V5 #129: Help, Y limit switch alarm shutting me down.

cdo1955, 2022-04-18:
When I first started using Xpro V5 the Y limit alarm would come in randomly. Mostly I could cut a project all the way through, but some times not before the alarm came in. The Y alarm shows on the screen using any of the G code senders, but the alarm light on the side of the controller never goes on. It happens constantly now and it never matters if the spindle is on or not. I'm wired with NO switches which I have verified are wired correctly. X and Z work perfectly. But Y alarm comes in all the time now. Without the Y limit switch alarm light ever coming on.
Below are my vitals:

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
**** Connected to COM3 @ 115200 baud ****
**** Connection closed ****
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
**** Connected to COM3 @ 115200 baud ****
[MSG:Grbl_ESP32 Ver 1.3a Date 20210816]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1250.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,1240.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,1240.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-100.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:No spindle]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
>>> $$
$0 = 4    (Step pulse time, microseconds)
$1 = 250    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 3    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 1    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 1000.000    (Homing search seek rate, mm/min)
$26 = 10.000    (Homing switch debounce delay, milliseconds)
[... 250 lines cut ...]
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.500
$Y/Current/Hold=0.500
$Z/Current/Hold=0.500
$A/Current/Hold=0.500
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
[Error] An error was detected while sending '$s': $Errors/Verbose=Off. Streaming has been paused.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
>>> ?
ok

Mawg1, 2022-04-19:
It sounds like you might be getting noise...maybe?   What is the response to the '?' command?  If you turn off 24V supply then send the '?' command what is the response?  

Generally, I would recommend NC switches as they much less susceptible to noise/interference  - even more so if you use a VFD

cdo1955, 2022-04-19:
Thanks for the response. Right now this Y limit alarm is coming on when the spindle is not even powered up.
I did try moving the wires to NC and changing the setting to invert limit switches but That did not fix the problem. So I returned to normal settings. 
I have 5 VDC measured across the terminal of the Y switches (wired in parallel) and when one  closes the voltage drops to zero. Which is correct since it is now connected to common. But the light on the side of the controller never show on. I only get the alarm on UGS. Can the Y circuit inside the controller be damaged?

cdo1955, 2022-04-19:
I did not mean to close this Issue, I just hit the wrong button.

cdo1955, 2022-04-19:
The response to the "?" with power supply on is in the list above. I will have to try to night testing "?" when the 24 VDC is off.

Spark-Concepts, 2022-05-16:
How are your limit switches wired? (are you wiring using all three terminals as illustrated below) 
- be sure to pay attention to the markings on your particular switches "NC" & "NO"
- you can also try moving the internal "Limit Switch Logic/Output Voltage Select Jumper" to 24V (this will decrease the limit switch trigger sensitivity"

[image]

cdo1955, 2022-05-17:
No, I was worried about you warning about doing it wrong and burn up the board.
I tried NO and NC. also 5 volt and 24 Volt option.
 X and Z work fine, never a moments problem with them.
 Y works for a while the comes in for no reason.

In the Three wire option you show on the  Gethub is ground thew actual earth ground from the circuit breaker board?
 carl Olsen

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F8650709%2F168652506-778a6c46-e999-45b0-9300-de90200fc6d7.png&data=05%7C01%7C%7C51cc95adf9cd4119ab0008da376ef65d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637883246956305442%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=K4pDUQA%2FD859c4KZKOzBlCCglbvrQmlBGWmW6207VLY%3D&reserved=0>

cdo1955, 2022-05-17:
The NC option shows the alarms on the panel toggle correctly when I set $5 to the correct setting. But when i try to home
it does not responed to the switch closure. And it would crash if I let it. When I set it back to NO with the jumper set to 24 VDC it does the exact same thing. The only way it works in with 5 VDC and NO. I can home and start to run a cut but sooner or latter Y alarm drifts in and shuts down the system. If I have already Homed the system I can kill hard limits and run on soft limits to get something done.
Is the Y circuit damaged in some way?
 Carl Olsen

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F8650709%2F168652506-778a6c46-e999-45b0-9300-de90200fc6d7.png&data=05%7C01%7C%7C51cc95adf9cd4119ab0008da376ef65d%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637883246956305442%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=K4pDUQA%2FD859c4KZKOzBlCCglbvrQmlBGWmW6207VLY%3D&reserved=0>

Spark-Concepts, 2022-05-17:
cdo1955, 

The internal ground is common within the board - you want to use the ground on the limit switch plug and not chassis or earth ground (don't daisy chain grounds either) otherwise you'll run into all kinds of issues with ground loops. 

Does the limit switch LED illuminate for each of the limit switches when triggered?

If you can - please post a picture of your y limit switch wiring as well as the wiring to your xProV5.

cdo1955, 2022-05-17:
I never see the limit switch indicator lights come on. The alarm indication I'm talking about in on UGS or CNCj's panels. As for the pictures I'll do it soon, , at work now.
 Carl

cdo1955, 2022-05-18:
[image]
[image]
Attached are two photos of the Y wiring. I use Two Limit switches on the Y normally like I do on the X and Z. They are wired in parallel. For testing I disconnected the wires on the far Y limit switch. But that why there are dual wires going to the switch. On the photo of the Y limit wires landing in the XPro V5 I pulled the Y motor plug to show the Y limit plug. I have tried it with a small cap across the input but now for trouble shooting I removed it.

cdo1955, 2022-05-18:
The shield wire are all landed on a bus bar tied to the power supply ground.

Spark-Concepts, 2022-05-20:
I see a couple options you could try:
- connect shields on the first image and connect the shield closest to the xPro to the bus bar (terminate shield to ground at a single point)
- grounds can be a tricky thing - do not daisy chain grounds (this will create ground loops and issues)
- if you are picking up noise you can try adding 1K resistor on the Y limit plug between the center pin (Vcc) and the signal pin

If you tried these things or are uncertain - contact us via: https://www.spark-concepts.com/contact-us-1/ and we'd be happy to send you a shipping label and test/repair/replace your unit free of charge

cdo1955, 2022-06-01:
I rewired all the limit switch's as per the diagram you gave me in this post. Things are working fine now . No limit switch problems. But I did have to set $5=0. If it set to $5-1 all the alarms are in. It homes fine with $5=0. Also I had to give up the limit switches I had at the opposite ends of all the axis. Because it looks like there's no why to have two limits on the same line wired in parallel or series. which is one of the reasons I went with the two wire system in parallel in the first place. I enabled soft limits and test them out. So if things say the same I'm happy with the out come.

cdo1955, 2022-06-06:
ZY alarm us back. I will contact you to send it back for repair

cdo1955, 2022-06-07:
Problem is back. Ill send it in for repair

cdo1955, 2022-06-22:
After I received the replacement controller my issues went away. I have had only one day cutting but normally that would have shut down due to Y limit, so far so good. I'll close this help request

cdo1955, 2022-06-22:
done

closed 2022-06-22 by cdo1955