Spark-Concepts/xPro-V5 #132: Unable to use to CNC xpro v5

szaghi, 2022-04-22:
Hi all, 

I am trying to use a CNC xpro v5 controller (bought for a queenbee pro CNC), but I am failing (probably due to my inexperience). The board starts correctly, it creates its own WLAN and I am able to connect both directly via USB cable and also by wifi connection to its WLAN. However, neither with CNCJS (via USB) nor with its own web GUI (via wifi) I am able to obtain _any response_ : the connection with the board seems to happen right, but I am not able to run any command (homing, jog, etc...), even the console seems to be unresponsive (sending $i or $S for example returns no answer). 

I am able to obtain a response only by pressing the physical reset button on the board: when I press the reset button I read in the console that the drivers are fine and the Z stepper tries to turn for a short while (with some noise), but after that I am still with a totally unresponsive board. 

I have updated the firmware to the last one (NC version).

Can someone help me to _debug_ this situation? Without communicating with the board I am not able to understand what is going on (e.g. if there is a wiring issue).

Thank you in advance for any suggestions,
Stefano

spydercnc, 2022-04-22:
Have you got a NC switch connected? If not it will see it as a door being open.  You can't connect to Wi-Fi with the USB plugged in. Which spindle are you using and which method are you using to control the spindle, the firmware is different for a RS485 connection.

szaghi, 2022-04-22:
> Have you got a NC switch connected? If not it will see it as a door being open.  You can't connect to Wi-Fi with the USB plugged in. Which spindle are you using and which method are you using to control the spindle, the firmware is different for a RS485 connection. I am sure I am using the NO firmware though?? 
> 

Hi @spydercnc , thanks for the quick support.

The limit switch should be correctly wired (over C and NC pins), tomorrow I can re-check the wiring and posting some pictures. 

The tests with USB and wifi connection were made separately, I have not connected USB and wifi together.

The spindle is a 2.2KW water cooled and I was trying to connect by RS485, maybe this is the problem.

The firmware update (made using web UI) was smooth, but as I said I am currently not able to obtain any response from the board 

In summary the first things I have to check are the wiring of limit switches and firmware for the spindle, right?

spydercnc, 2022-04-22:
Not the limit switches that may be the issue but the door switch and emergency switch needs to be NO.  To use the RS 485 see this: Setting up the CNC xPRO V5
1) Install the latest firmware
a. For Normally Open Door Switches (or no door switch): xProV5/CNCxPROv5_XYYZ_NO.bin at main · Spark-Concepts/xPro-V5 (github.com)
b. For Normally Closed Door Switches: xPro-V5/CNCxPROv5_XYYZ_NC.bin at main · SparkConcepts/xPro-V5 (github.com)
c. Load using these directions: Checking_firmware_and_upgrading · Spark-Concepts/xProV5 Wiki (github.com)

spydercnc, 2022-04-22:
[CNC.xPRO.V5.RS485.control.for.HuanYang.VFD.Spindle.pdf](https://github.com/Spark-Concepts/xPro-V5/files/8543206/CNC.xPRO.V5.RS485.control.for.HuanYang.VFD.Spindle.pdf)

szaghi, 2022-04-22:
Oh sorry for my misunderstanding, I was thinking to limit switches, not to the door and emergency stop, tomorrow I'll check them and I'll fix the firmware.

Thank you very much.

szaghi, 2022-04-23:
@spydercnc Thank you very much for the support, I found my error: I assumed to have NO emergency stop, but today I re-opened it and I found it is a NC. Flashing the CNCxPROv5_XYYZ_NC.bin firrmware I am now able to dialog with the board. 

For now all seems to work as expected except for the second Y stepper: it seems there is a "delay" between the two Y steppers, one starts before the other. I have to dig it deeper, anyhow I am closing this issue (if I'll not be able to understand what is happening with the Y steppers I'll look for help).

Thank you again,
Stefano

closed 2022-04-23 by szaghi