Spark-Concepts/xPro-V5 #133: Problem with the Y-axis (x pro-V5)

M-a-K-A-R-o-N-k-A, 2022-04-24:
Hello!!!
There is a problem with the x pro-V5, namely on the Y axis the engine behaves incorrectly.
After checking the engine on the other axes X,Z,A/Y2. It turned out that it is serviceable and works normally.
What else do I need to provide you to speed up the resolution of this issue?
Ps: I'm new to this business
[Y problem](https://drive.google.com/file/d/11ggokg1jSniMxvOAmu_hxbrkAmdxh9ZA/view?usp=drivesdk)
> $I
[VER:1.3a.20210816:]
[OPT:MPHBSW]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Mode=AP:SSDI=CNC_xPRO_V5:IP=192.168.0.1:MAC=40-F5-20-B5-67-F5]
[MSG:No BT]
ok

> $$
$0=4 (Step pulse time, microseconds)
$1=250 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2500.000 (X-axis maximum rate, mm/min)
$111=2500.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=50.000 (X-axis acceleration, mm/sec^2)
$121=50.000 (Y-axis acceleration, mm/sec^2)
$122=50.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=300.000 (X-axis maximum travel, millimeters)
$131=300.000 (Y-axis maximum travel, millimeters)
$132=300.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok

> $+
$0=4 (Step pulse time, microseconds)
$1=250 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2500.000 (X-axis maximum rate, mm/min)
$111=2500.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=50.000 (X-axis acceleration, mm/sec^2)
$121=50.000 (Y-axis acceleration, mm/sec^2)
$122=50.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=300.000 (X-axis maximum travel, millimeters)
$131=300.000 (Y-axis maximum travel, millimeters)
$132=300.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
$140=1.800
$141=1.800
$142=1.800
$143=1.800
$144=0.250
$145=0.250
$150=0.500
$151=0.500
$152=0.500
$153=0.500
$154=0.125
$155=0.125
$160=8
$161=8
$162=8
$163=16
$164=16
$165=16
$170=16
$171=16
$172=16
$173=16
$174=16
$175=16
ok

szaghi, 2022-04-24:
Hi @M-a-K-A-R-o-N-k-A , I think I have the same issue, thus I would like to follow this issue.

possu911, 2022-05-02:
Hi,
Mine just started to do the same with Y1 (today May 1’ 22)
Received the xPro early Apr.
Stock Firmware.

JosDrolkoppers, 2022-05-27:
Hi, I have the exact same issue. I already purchased a new controller and I got the same problem after 30 minutes of running...
video : https://imgur.com/gallery/vNHIggC

Kind regards,
Maarten

Fcsteve777, 2022-05-31:
Spark concepts sent me another unit, hopefully this isn't a recurring issue. I did inquire if they know what is causing these issues.

Fcsteve777, 2022-06-03:
well same as Jos up there, got my replacement and a half hour later back to effed up....

shnads, 2022-06-11:
I am also having the same issue, purchased my Xpro V5 around two weeks ago, managed to get around two hours of use while setting up my machine, came back the next day and it has this problem. Beginning to regret going for the xPro as I can't just swap out a stepper driver, i now have to wait for replacement.

Spark-Concepts, 2022-08-09:
We are looking into several of these occurrences and are actively working with Trinamic to determine the root cause of this fault.  That said we are confident on an expedient resolution - firmware and/or hardware.

That said, this type of issue is still relatively rare in most cases:  

Regardless, we are actively and rapidly working to determine a definitive solution and will make good on providing you a quality product.
 
We understand this may be extremely frustrating to all - in the meantime if you're seeing this issue with the Y axis specifically (and not Y2) please utilize the contact us  link on the spark-concepts.com homepage and let us know when and through who you purchased your xPro from (we're exploring all avenues as to why/how this issue is occurring)

In most all cases we can provide you a replacement - however, if at all possible I highly recommend an alternative (temporary) solution until we have a clear answer and resolution to this rare but frustrating occurrence:
In the meantime to resume use of your CNC you can run both Y steppers via the Y2 stepper output either in Series (higher torque but slower feed rates) or in parallel (lower torque but faster feed rates).

Regardless, if your having this specific issue please reach out to us directly via our [website](www.spark-concepts.com) Contact-Us tab and provide us as much detail as you can and we will reach back out to you with options

closed 2022-08-09 by Spark-Concepts