Spark-Concepts/xPro-V5 #134: xPro V5 Wifi problem

Monya0065, 2022-04-26:
When 24v power is on, there is no Wifi connection. Wifi turns on only when connected to a computer. How to fix it?

Spark-Concepts, 2022-05-16:
It sounds like you may have had something external cause a short circuit which can sometimes damage the xPro's 5V regulator.

If possible - can you run a few tests:

1. Disconnect the USB cable and power only using your 24V power supply
2. Remove the lid by removing the four lid screws and expose the top PCB
3. Turn your 24V power supply ON 
4. Measure for 5 Volts as shown below (also check the remaining voltages as shown on the right)
5. Remove 24V or turn OFF 24V power supply
6. Plug in USB cable, and power via a PC or USB power brick 
7. Re-Measure for 5V and 3.3V (24V should be 0-ish)
8. Please let me know your test results (actual voltages measured)

| Voltage  | 24V PSU Only | USB Only |
| ------------- | ------------- | ------------- |
| 3.3V  | (value?)  | (value?) |
| 5V  | (value?)  | (value?) |
| 24V  | (value?)  | (value?) |

[image]

Spark-Concepts, 2022-08-09:
Has this issue been resolved?

closed 2022-08-09 by Spark-Concepts