Spark-Concepts/xPro-V5 #135: The board is not connected to the computer (CNCjs, Mach3 and other)

Monya0065, 2022-04-26:
The board stopped connecting to the computer. When connecting USB-C, CNCjs tries to connect but fails. Other programs do not determine the fee. Drivers are up to date. WiFi cannot be detected and connected.
How to fix it?

Spark-Concepts, 2022-05-16:
Do any of the LED's illuminate? 

The first thing to look at would be power - if possible can you run a few tests found here: https://github.com/Spark-Concepts/xPro-V5/issues/134#issuecomment-1127978054

BungC, 2022-05-21:
I'm having similar problems with connecting, with CNCjs 1.9.23. The steppers "click" and connection appears to be ok, but I cannot issue any commands or get to "idle" state to be able to jog, etc. Universal G-code sender will talk, but only after a few resets of of both UGS or xPRO or both. CNC3D Commander (an ESP32 ?GRBL "1.3" sender" also behaves strangely and requires  a reset of the xPRO to work. I will try loading the new firmware, the current firmware is from September 2021.

Spark-Concepts, 2022-08-09:
Are you still having issues? - if not please repost. I'm assuming firmware should've fixed the reset issue.

closed 2022-08-09 by Spark-Concepts