Spark-Concepts/xPro-V5 #137: Help me before I destrod my machen :(

ghalib1399, 2022-04-30:
Hi all I have problem to connect xpro v5 with VFD, and I tried many option but does not work with me!(Issues #109 ).
  if your (cnc machine) is working smoothly, please share a visual representation of how you connected xpro v5 with VFD .

spydercnc, 2022-05-01:
Try and search for my name on here. I posted the link to get the RS485 working, two wires and the correct firmware on the XPro and around 6 lines changed on the VFD. I will try and look for it in the morning. Easy when it all falls into place.🙂

ghalib1399, 2022-05-01:
> Try and search for my name on here. I posted the link to get the RS485 working, two wires and the correct firmware on the XPro and around 6 lines changed on the VFD. I will try and look for it in the morning. Easy when it all falls into place.🙂

Thank You very much 
You guided me to solve my problem !!! it was software not hardware ! I install old firmware(CNC_xPRO_V5_XYYZ_485_NC) and work fine

closed 2022-05-01 by ghalib1399