Spark-Concepts/xPro-V5 #140: Y axis issue stepper is hissing and more

Pops1971, 2022-05-07:
After tuning the first short test cut. One of the y steppers started hissing and making video game noises. It’s also stuttering and jamming up the machine. It was running fine before the test cut. Box is not showing any lights. The stepper does not seem to get warm either. I have noticed the fan on the pro also does not work right. I have updated the firmware and have checked all setting and made sure machine is square. It’s barely 2 months old and I just got it hooked up over the last few days. I’m out of ideals and losing work. I’m a disabled person and sold my little cnc to get this bigger one. Any help would be greatly appreciated. Thank you

spydercnc, 2022-05-07:
Check that the wires are paired correctly and that the stepper is running in the correct direction. Make sure the axis is not jamming (unlikely)

Pops1971, 2022-05-08:
One thing I noticed is my power supply is only putting out 22v. Could this be causing some of the stepper issues?

Spark-Concepts, 2022-08-09:
Low voltage may effect it some, however I would suggest updating to the most current firmware as of July 2022 - If you are still having issues please reach out to us via our contact us link on our homepage www.spark-concepts.com

closed 2022-08-09 by Spark-Concepts