Spark-Concepts/xPro-V5 #141: Spindle/VFD communication error

NZBasher, 2022-05-10:
Hi, I have tried everything to get my spindle to run via RS485. This worked initially but has now stopped.
I note I do not get the same message on bootup as others I have seen on here
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]

I notice I have no 
[MSG:Auto-RTS in use]

After going back a firmware version, I can get the spindle to start with m3 command, an error then comes up but I can adjust speed and stop it with M5 (CNCjs used) If I try to restart the spindle starts to spool up and stops with unresponsive 6 etc. error shown below. I have to reset the spindle and VFD to get it to work again

I'm getting totally frustrated, any help appreciated.
Thank you
Paul

feeder> M3 S10000
ok
[MSG:Spindle RS485 Unresponsive 7]
feeder> M5
ok
feeder> M3 S10000
ok
[MSG:Spindle RS485 Unresponsive 6]
[MSG:Critical Spindle RS485 Unresponsive]
ALARM:10

other settings below for your reference

$0=4 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=3 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=8000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$100=400.000 (X-axis travel resolution, step/mm)
$101=400.000 (Y-axis travel resolution, step/mm)
$102=400.000 (Z-axis travel resolution, step/mm)
$103=400.000
$104=100.000
$105=100.000
$110=3000.000 (X-axis maximum rate, mm/min)
$111=3000.000 (Y-axis maximum rate, mm/min)
$112=3000.000 (Z-axis maximum rate, mm/min)
$113=3000.000
$114=1000.000<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-780.322,-1231.843,-84.802>
$115=1000.000
$120=250.000 (X-axis acceleration, mm/sec^2)
$121=250.000 (Y-axis acceleration, mm/sec^2)
$122=300.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=1220.000 (X-axis maximum travel, millimeters)
$131=1270.000 (Y-axis maximum travel, millimeters)
$132=120.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(-1220.000,0.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(-1270.000,0.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(-1270.000,0.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-120.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(21)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
> $soindle
error:3 (Invalid statement)
> $spindle
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Spindle/Delay/SpinDown=20.000
$Spindle/Delay/SpinUp=20.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=HUANYANG

NZBasher, 2022-05-11:
Help please

Spark-Concepts, 2022-05-20:
Please take a look at the updated wiring guidance found here:
https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#spindletypexx--rs485

_(most all VFD communications errors are due to noise emitted by the VFD/Spindle; therefore, special wiring considerations need to be taken to help mitigate communication errors)_

id192313, 2022-06-02:
Hi, did you manage to get it solved?

NZBasher, 2022-06-03:
Hi,

Thanks for getting in touch,

I haven’t tried to get the RS485 going again yet. I have the 2 resisters and some shielded twisted pair and will change firmware and try it this weekend.

I’m currently setting speed on the vfd and using the relay for start/stop

I have tried PWM which works fine until I start moving the machine and I get varying voltages which causes the spindle to speed up and slow down constantly. There seems to be a lot of voltage variation. I have had a multi-meter on it and its definitely varying, I note that it speeds up when I touch the voltage sensor to the contacts! Seems a bit stramge, I can only get about 5.7 volts when the spindle is connected but get 12 when its not. So not sure what’s going on there, but its unusable.

I’m not sure if the PWM and the RS485 issues are connected. The silly thing is that it all worked fine when I first assembled the machine, it wasn’t till my electrician found my shed wasn’t earth bonded so the RCD system wasn’t working (bit dangerous) after that RS485 wasn’t working, even putting the earth back as it was didn’t work, so not sure if that was just a coincidence.

 

I’ll let you know how I get on.

Regards

 

Paul McBreen

id192313, 2022-06-03:
Hi Paul, 
Thanks for replying. Been working on it for the past 2 days and was wondering how u did the connection using PWM. I have tried PWM setting today. Can’t even get the spindle to start. 
Is these how u set it up previously?

-xPro v5, HY02D223B, Spindle 2.2kw 220v Air Cooled 
-VFD setting same as the one mactec54 posted.
-xpro Switch EN/PWM
-xpro 0-10v to VFD VI
 xpro GND to VFD ACM
 xpro Relay NO to VFD DCM
 xpro Relay GND/COM to VFD FOR
-xpro Relay jumper Spindle Enable/SP_EN (the relay kept on and off the whole time)

Thank you.

Spark-Concepts, 2022-08-09:
Have you been able to resolve this issue?

https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453

closed 2022-08-09 by Spark-Concepts

NZBasher, 2022-08-09:
Not yet,

I have been using the relay for start stop of the spindle and manually setting speed. I will try the new method as soon as I get time.

I tried using PWM 0-10v but the voltage is inconsistent so the spindle speeds up and slows down, making it unusable. I’m not sure my Xpro is working correctly

 

Regards

Paul McBreen

 

#164 (comment) <https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453>

Spark-Concepts, 2022-08-09:
Sounds good Paul - take a look at Issue #164 - and please feel free to reach back out if you sill aren't having any luck. I'm assuming you're running a 220-240V VFD?

NZBasher, 2022-08-09:
Yes 220volt, we have a very clean supply here as well, mix of solar and oversized grid supply due to allowance for future expansion of housing.

Regards

Paul McBreen