Spark-Concepts/xPro-V5 #142: New install.  All stepper motors error

spidee601, 2022-05-21:
Ok, I just assembled my Ultimate Bee 1500x1500 ball screw.  It has 4 Nema23 steppers.  I updated the firmware to CNC_xPRO_V5_XYYZ_BM3D_NC because I have the emergency stop button. When I try to move any of the steppers, they make a "clicking" sound then the xPro error light turn on for that stepper.  I've tried several different things I've read on other threads but nothing seems to work.  Below are my specs, I know I'll have to change some of the perimeters for my bed size, but it's one step at time. They don't move at all over wifi, but they respond with the click when using CNCJS. 

 Grbl 1.3a ['$' for help]
> $$
$0=4 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=12000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2500.000 (X-axis maximum rate, mm/min)
$111=2500.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=100.000 (X-axis acceleration, mm/sec^2)
$121=100.000 (Y-axis acceleration, mm/sec^2)
$122=100.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=300.000 (X-axis maximum travel, millimeters)
$131=500.000 (Y-axis maximum travel, millimeters)
$132=500.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000<Idle|MPos:0.000,0.000,0.000|FS:0,0>

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20220403]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_BM3D_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,300.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,300.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:No spindle]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]

I would appreciate any assistance in getting this up and running.  

Thanks in advance!

Spark-Concepts, 2022-05-22:
Make sure your emergency stop switch is pulled out (and verify it's configured Normally Closed when pulled out).

Also, please copy-paste a ```$S``` vs a ```$$``` so we can look at all of your settings - thanks!

spidee601, 2022-05-22:
Thanks for the response.  Yes, the E-button is pulled out

> $S
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=xPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.500
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=NONE
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2500.000
$Y/MaxRate=2500.000
$Z/MaxRate=2500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=100.000
$Y/Acceleration=100.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=300.000
$Y/MaxTravel=300.000
$Z/MaxTravel=300.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok

spidee601, 2022-05-22:
[image]

Ok I think I found the issue.  I wired them as directed on your diagram, but the wiring should be Red, Green, Blue, Yellow.  Once I did that the stepper motors fire up, grant it I'll have to figure out the right settings for my ball screw drive, but they seem to be working now.  

Since you have all of my settings any advice on what the other settings should be on a 1500x1500 ball screw Ultimate Bee?