Spark-Concepts/xPro-V5 #144: Y driver issue

Fcsteve777, 2022-05-23:
Hello,
My machine was running swimmingly over a couple of days then 2 issues started which now has me dead in the water after finally getting everything ready to go. 

Although i did not change any settings my X axis stepper was becoming very hot ~120c before i stopped the machine. I re checked my amp limits which were still where i left them.

The other issue is that one of the y drivers seems faulty and now the stepper is making massive amounts of noise when on the machine. I swapped motors and drivers and the noise always follows the driver. it started after running an hour or so the other day but on startup it was doing the same thing. 

Videos:
https://drive.google.com/file/d/1HnIgDp35-XVLFc52DLajtfe9ngFjktYK/view?usp=sharing
https://drive.google.com/file/d/1Hpwh1PZrVfcIXKb0-OjnDbcs6bZaz7d7/view?usp=sharing

Today i updated the firmware as i saw some driver tuning was noted but there was no change.

I tried slowing my speeds and the frequency just changes, no error lights illuminate

Driver self test results:
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,508.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,782.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,782.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-140.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]

Settings are:
$0=4
$1=250
$2=0
$3=6
$4=0
$5=1
$6=1
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=3
$24=200.000
$25=2000.000
$26=250.000
$27=2.500
$30=24000.000
$31=8000.000
$32=0
$100=159.860
$101=160.150
$102=401.600
$103=200.000
$104=100.000
$105=100.000
$110=3000.000
$111=3000.000
$112=2250.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=80.000
$121=80.000
$122=50.000
$123=200.000
$124=200.000
$125=200.000
$130=508.000
$131=782.000
$132=140.000
$133=300.000
$134=300.000
$135=300.000

Fcsteve777, 2022-06-03:
Update,
Received replacement unit and installed with the same settings. 

Ran great for a half hour then noticed one of my y steppers stopped working. Everything ohmed out on the motors between 1.6 and 1.8 ohms. After re setting everything its back to buzzing the hell out of one of the steppers....

This is a 230$ letdown so far.

Spark-Concepts, 2022-08-09:
We are actively working with Trinamic to determine the root cause of these issues - please use the contact-us link on the Spark-concepts.com site for resolution

closed 2022-08-09 by Spark-Concepts