Spark-Concepts/xPro-V5 #145: xPro-V5 not responsive

RidwaanN, 2022-05-24:
Hi I have my xPro-V5 just under 2months now, it's been working fine all the time.
I last cut 2 weeks ago Friday. Monday passed i switched everything on as normal. Plugged in the controller via USB-C cable as usual and opened CNCJs.
As usual I clicked on Reset and Unlock. Where i would usually get all my configurations printed out on the left column - Nothing. The port is open, xPro V5 is powered on (Green light is on and blue usb light is on). It wont unlock or reset. Common response to anything is $X
I have tried closing the port and CNCJs, even restarting laptop and also tried connecting via LaserGRBL which will not connect successfully.

[image]

RidwaanN, 2022-05-25:
Hi. It appears to be in my device manager...
[image]

RidwaanN, 2022-05-25:
[image]
[image]

Mboss32, 2022-05-27:
Try opening the port in CNCjs, then pressing the "reset" button on the side of the controller

RidwaanN, 2022-05-28:
> Try opening the port in CNCjs, then pressing the "reset" button on the side of the controller

I have tried that. I even held it in for +/- 20 seconds with no response.

Jennywhite2, 2022-07-16:
> > Try opening the port in CNCjs, then pressing the "reset" button on the side of the controller
> 
> I have tried that. I even held it in for +/- 20 seconds with no response.

Hello RidwaanN,may i know,where u come from,,malaysia?

Spark-Concepts, 2022-08-09:
Has this issue been resolved?

closed 2022-08-09 by Spark-Concepts

Jennywhite2, 2022-08-26:
[image]

Jennywhite2, 2022-08-26:
[image]