Spark-Concepts/xPro-V5 #146: Y  Axis Driver Short Coil

manicresin, 2022-05-27:
Machine was working fine for a few days and I progressed through my build.
Yesterday the Y-Axis stopped working.

I disconnected all terminal blocks except for power and used a known good motor only connected to the Y axis with no change.
The same motor work on X and Z axis without issues.
I upgraded to the latest firmware also with no change.

```
>>> $g
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20220403]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NO]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Hold switch on pin GPIO(13)]
[MSG:Start switch on pin GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,300.000)]
[MSG:Y  Axis Driver Short Coil a:Y b:N]
[MSG:Y  Axis Driver Short vsa:Y vsb:N]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,300.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:No spindle]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
>>> $+
$0 = 4    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 1    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 2.500    (Homing switch pull-off distance, millimeters)
$30 = 24000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$N1=
$N0=
$33 = 5000.000   
$34 = 0.000   
$35 = 0.000   
$36 = 100.000   
$100 = 200.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$103 = 200.000   
$104 = 100.000   
$105 = 100.000   
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 2500.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 300.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
$140 = 1.800   
$141 = 1.800   
$142 = 1.800   
$143 = 1.800   
$144 = 0.800   
$145 = 0.800   
$150 = 0.500   
$151 = 0.500   
$152 = 0.500   
$153 = 0.500   
$154 = 0.400   
$155 = 0.400   
$160 = 8   
$161 = 8   
$162 = 8   
$163 = 16   
$164 = 16   
$165 = 16   
$170 = 16   
$171 = 16   
$172 = 16   
$173 = 16   
$174 = 16   
$175 = 16   
ok
```

jaybay-77, 2022-06-02:
I've just sent mine back to makerstore here in Australia where I bought it. We did some remote troubleshooting and it definitely looks like an issue with the Y port. They are going to test it and replace if it's faulty. This seems to be fairly common issue and it's pretty slack that spark don't monitor and repond to their own customers.

Spark-Concepts, 2022-08-09:
jaybay, 

Our customer interaction is handled directly via our contact us via our homepage Spark-Concepts.com - we are happy to assist. If you attempted to use the contact us and don't get a response within 1-2 business days please try again as on occasion some emails get sent directly to spam for "content"; astounding the number of pornographic and explicit spam emails on a daily basis that our spam filter blocks; unfortunately sometimes that includes legit emails too. 

Regarding the Y axis issue - albeit somewhat relatively rare we are actively working with Trinamic to determine the root cause of the uptick with this particular issue.  We are confident in an expedient resolution. We take the upmost pride in our product and will stand by it.  

Again, if you are having the specific issue in that the Y axis (specifically) is faulting, I urge you use the contact us link on our homepage and provide us as much detail at to the issue your seeing and when/and through whom you purchased your xPro from - we're exploring all avenues from firmware to hardware and manufacture/supply chain.

closed 2022-08-09 by Spark-Concepts

manicresin, 2022-08-09:
It was functioning normally then it was powered off and when powered back on it was not functioning at all and reported and error on that driver.

Spark-Concepts, 2022-08-09:
Thanks Manicresin,  have you recievd a replacement unit?

jaybay-77, 2022-08-09:
Same here, unit was working fine but then when powered up to run a job the error occurred.

manicresin, 2022-08-11:
> Thanks Manicresin, have you recievd a replacement unit?

No, I was ask for shipping details but never received a replacement or further instructions.