Spark-Concepts/xPro-V5 #147: Xpro v5 y axis fault

jaybay-77, 2022-06-01:
As of yesterday my y axis stepper fault indicator is on. The controller has been fine until my second job yesterday. The y axis now doesn't run correctly, it makes an awful grinding sound and doesn't move properly. If I switch the y2 motor over to the y port it also does the same thing, and the y motor runs fine when connected to the y2 port. To me this indicates a fault in the controller. Please have a look at the video and grbl settings and let me know how to fix this. It's only a year old and I am now unable to do anything until I get this resolved. 
[Video](https://photos.app.goo.gl/g99Kdnhw7dHCTGkA9) [Grbl settings](https://drive.google.com/file/d/1-a6slgBp1NLDKgP5pNwxNvG62gNtUdYH/view?usp=drivesdk)

jaybay-77, 2022-06-02:
This has been sent back to Makerstore here in Australia for a fault test. Connecting the y motors directly to the controller proved that there is an issue with the y port:[Running a test grbl fiule form maker store on just the y axis](https://photos.app.goo.gl/SEkFU2SZrSgpYLmr7)

Appreciate an update from Spark on this issue as it seems to be very common. Maybe faulty hardware in the controller which they should be advising us about???

jaybay-77, 2022-06-08:
This unit was deemed faulty and replaced by Makerstor. Spark Concepts obviously don't bother checking their issues list so this is more for any users who may come across this issue.

Spark-Concepts, 2022-08-09:
We are actively working with Trinamic to determine the root cause of these type of issues - Please use the contact-us link on the Spark-concepts.com site for resolution

closed 2022-08-09 by Spark-Concepts

danatomy, 2022-11-22:
I have the same issue. Have contacted sparks.