Spark-Concepts/xPro-V5 #149: G38.2 probe causes lockup in macros

NeonHorizon, 2022-06-03:
Hardware: CNC xPRO V5
ESP3D-WebUI: 2.1b52 
GRBL-ESP32:  1.3a
Firmware: CNCxPROv5_XYYZ_NC.bin (20220403)

If I enter the following commands in the WebUI interface they work perfectly every time:
```
G10 L2 X0 Y0 Z0      ; Set the working coordinates to match the machine coordinates
G90                  ; Use absolute positioning
G38.2 Z7 F50         ; Probe Z, dont go closer than 7mm from Z0 as the touch plate is 7mm
G91                  ; Use relative movements
G0 Z1                ; Clear the touch plate by 1mm
```

The mill descends, touches the plate and backs off as expected.

However when the exact same code is placed in a WebUI macro, the probe moves downwards at 50mm/min, touches the plate then CONTINUES to descend but at a much much slower rate indefinitely pushing the bit into the touch plate and ignoring the gcode placed after the G38.2. The WebUI interface also becomes non responsive and the XPro has to be power cycled to be usable again.

The probe has been checked and is functioning perfectly.

UPDATE 1 - I've just been testing and noticed exactly the same behaviour can be observed using the following simple one liner in a macro:
```G38.2 Z5.0 F50```
There's a 50mm/min descent and instead of stopping after touching it drives the bit hard into the touch plate at a very low speed.

UPDATE 2 - I just tried the following code in a macro:
```
G91
G38.2 Y5 F50
G0 Z1
```
It moves towards the touch plate at 50mm/m on the Y axis as expected, touches the plate then continues to move at a very slow speed and again WebUI is locked up until a power cycle.

It appears as if G38.2 has issues when used in WebUI macros?
At least on my set up?

Thanks

NeonHorizon, 2022-06-08:
Just an update on this, if you upload the code to the SD card and run it as if its a regular job it works fine. So its literally a macro issue.

haksko, 2023-02-07:
I have the same problem. Is there a solition to this?