Spark-Concepts/xPro-V5 #150: Holding current issue

nyarurato, 2022-06-03:
I try new firmware "April 2022 Firmware".
It seems holding current is strange.
Below is the table of measurement result for holding current on $140=1.8. It applied after $RST=*.
I think the holding current is DC current, so I measured the current at an angle where the current flows through the coil on one side only.

|Parameter|Holding current (A)|
|---|---|
|150=0.1|0.2|
|150=0.3|0.6|
|150=0.5|1.0|
|150=0.7|1.3|
|150=1.0|1.7|
|150=1.2|2.1|
|150=1.5|2.8|
|150=1.7|3.1|

This result indicates that holding current setting may have been changed to a percentage setting of drive current setting.
I have checked the firmware source code and there does not appear to be anything wrong. However, it is old source code so I am not sure if it is correct now. (If anyone from SparkConcepts sees this, please update the source code! I'd like to check it )
https://github.com/Spark-Concepts/xPro-V5-Firmware/blob/main/CNCxPROv5_20210708/src/Motors/TrinamicDriver.cpp#L196-L215

Spark-Concepts, 2022-08-09:
We are actively working with Trinamic to figure out what's going on with this - definitely strange behavior

Spark-Concepts, 2022-08-09:
Please resume issue here: https://github.com/Spark-Concepts/xPro-V5/issues/121#issuecomment-1208771628

I will post updates once I have a more clear answer

closed 2022-08-09 by Spark-Concepts