Spark-Concepts/xPro-V5 #151: Connect Xpro V5 to wifi network

jaybay-77, 2022-06-08:
Given there is ZERO documentation on doing this, can **Spark** please provide some clear instructions on how to do this?

Spark-Concepts, 2022-07-12:
Default Settings
The first time you power up the ESP32 with Grbl_ESP32 it cannot connect to your WiFi network because it does not know the SSID and password for your network. It will a create its own Wifi access point with the following settings.

- SSID: GRBL_ESP
- Password: 12345678
- IP Address: 192.168.0.1

Connect your PC, Tablet, etc to this network. Most computers will automatically launch a browser to connect to Grbl_ESP32 after connecting via Wifi. If it does not, simply enter http://192.168.0.1/ for the address in your browser.

This is very convenient if you are just getting started or there is no existing network to connect to. Unfortunately, this typically causes your computer to lose its Internet connection, so you probably want to setup Grbl_ESP32 to use your WiFi network.

# Method 1: Using the WebUi

Generally, we recommend using the Access Point just for network security purposes.  If you can accept that risk, or have a local network you can change the behavior of the wifi connection with the below settings.  

[image]

The first two are hopefully obvious.  For the Station IP Mode, choosing a static IP will keep the xPRO from being re-assigned so select a Station Static IP that you know is on your network and free.  Add your gateway address (192.168.1.1 is fairly typical) and add the network mask (255.255.255.0 is also fairly typical).  

Make sure you click the "SET" button for each of these entries before moving on.  

Lastly, change your Radio Mode to Station.  

[image]

Click "Set".  Reboot the controller and connect your computer to your network, open a browser, and type in the assigned static IP.  

_note: if you need to reset the network settings back to default: Send command `"$Radio/Mode=AP"` through the command terminal in CNCjs and restart the controller to reset the Wifi to Access Point mode_

# Method 2: Using a serial terminal

Send the following command through the command terminal in CNCjs and restart the controller

`$Sta/SSID=YourSSID`
`$Sta/Password=YourPassword`
`$Radio/Mode=STA`

markcagatandavis, 2022-11-08:
This doesn't do anything..

jaybay-77, 2022-11-08:
Agreed. I gave up on this after following these instructions and getting nowhere.

markcagatandavis, 2022-11-08:
> Agreed. I gave up on this after following these instructions and getting nowhere.

I managed to get it working after 3-hours... If you want to give it a try:

1. Set your IP configurations using the WebUI is fine (Station Static IP, Station Static Gateway, Station Static Mask).
2. Don't use the webUI for the bellow, it doesn't work\save your settings.
3. Connect using USB-A to USB-C cable
4. In CNCJS console:

$Sta/SSID=Your WiFi SSID Name (case sensitive I found)
$Sta/Password=Your WiFi password
$Radio/Mode=STA

NOTE: For some stupid reason, when you add a value to the $Sta/ you can only add one at a time, and then you need to reboot the xPro in between each. Otherwise, the value doesn't save.

Once all the values have been done and rebooted in-between each configuration, make sure you power off the xPro from the wall. Since it uses a sperate power supply, it must keep the capacitors live or something? Give it 5min and authentication should work, and you can connect to the IP from your normal network.

I tested this a few times to see if I just needed a single power drain, but it seems to be required after setting each value. I also noticed when you enter the values via console, it doesn't give you an OK to let you know it's done... this is why a reboot is required after each value has been set.

markcagatandavis, 2022-11-09:
UPDATE: I wouldn't bother with WiFi... Turns out it's pretty unstable and drops out a lot. I am pretty much next to the router, so it would either be EMI or the wifi controller/driver is not great.  :)