Spark-Concepts/xPro-V5 #154: More VFD issues after firmware update. :(

spydercnc, 2022-06-17:
Using xPro5 with a 15 water-cooled VFD. Got the system working in January via RS485 and it has worked well, seamless for the most part actually.

spydercnc, 2022-06-17:
Update and a shoutout for CNC3D Commander. Used their update tool, downgraded the firmware to the previous RS485 NO.Bin file, added the $spindle /type and a little green light started flashing  with everything springing into life. Life can be simple at times-begs the question why it had to be complicated then??

Spark-Concepts, 2022-08-09:
Try installing the latest firmware as of July 2022 - If you're still having issues read https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453

closed 2022-08-09 by Spark-Concepts