Spark-Concepts/xPro-V5 #155: Can a X proV5 firmware be set to Grbl 1.1  ?

ice-tii, 2022-06-20:
Hello there!

I have an problem of comunicating between Picsender and my xpro V5.
I found out that grbl1.1 is the highest compatibel version for picsender,
so now i want to flash grbl 1.1 on it.

Is it possible and if yes, what would be the best way?

Thank you guys!

 Timo from Austria!

lagnat, 2022-06-24:
You might look into https://github.com/bdring/FluidNC, however stick with v3.4.4 for the time being.

Spark-Concepts, 2022-08-09:
No unfortunately GRBL1.1 is ported to a Atmel 328 processor and the xProV5 is a ported version based on GRBL 1.1 to an ESP32 processor module (with expanded capability)

I have several xPro V3's and 4's that are true GRBL 1.1 you could have for free if you pay shipping from the US to Austria lol

Do you currently have an xPro V5? It should work, however, I can't confirm without testing. Was there a specific reason you were looking at Picsender? There are many other sender programs that are very good like universal G-code sender and CNCJS

closed 2022-08-09 by Spark-Concepts

ice-tii, 2022-08-09:
Hello there!

 

Well that answer shure took a while :-)

 

I bought a v5 from Ratrig.com and i will send it back because the V+ 10V for spindel control raises and lowers the Voltage when i do a job.

But everything else works fine.

 

With Picsender you can go up to 2 millions lines, so for my topo Lanscapes, which sometimes go in that direktion,

picsender really does the job, without having issuses. 

At first i worked with ugs and after several hours, the z axis, plunged suddendly in the material and ruined my router and the HDU Material.

Then i came across a guy from the us which said that picsender is specially for big files and now that i work with that i am so happy and no problem at all.

 

Since i want to do Workshops with young and old people, that they can get in touch with Cnc and learn step by step on how to do that,

i would take your offer to ship the v3 v4.

 

In that case could you find out what shipping would cost? 

Here in my area we dont rally work with grlb, more with mach3 and i want to introduce people in the world of grbl and the workbee cnc kit.

 

Thank you for your answer!

 

Mit freundlichen grüßen

Timo Helder
3D WELTEN
Ö/Vorarlberg
6943 Riefensberg
Riebinger 136

+43 6763632706
www.3d-welten.at

ATU74032806

 
 

Gesendet: Dienstag, 09. August 2022 um 04:54 Uhr

Spark-Concepts, 2022-08-09:
Timo,

I'm the one from Spark-Concepts who monitors the github - unfortunately back in early June I had to tend to a family emergency and was unable to give github the love it requires. All is good now, however, I'm catching up on answering questions.

First off regarding xPro's PWM output raising and lowering, I've heard a couple folks having a similar issue. My hunch is that it may be a grounding issue on your VFD (most are). Especially if you've tried running your VFD over RS485 with similar issues. 

Since we're based in the US, we've only been able to fully test the 110V VFD's using the xPro - we really haven't had any issues like what you're seeing (either it works or it doesn't)...  oddly enough, 98% of folks reporting VFD issues are running 220-240V VFD's. That said, we recently purchased a 110 to 240 voltage converter and are awaiting delivery of a 220/240V VFD - you'd think they'd be relatively similar in design, apparently they are different animals.  That said I plan to evaluate the VFD design and look for any design aspects that may be influencing issues like what you are seeing. Please keep me in the loop as to the status of your VFD controller once your new controller arrives.

Now, regarding the V3's and V4's... how many V3's do you want?, I found a box full of them - I'm not sure I if I have any V4's. Regardless, I will have Mike take a look at cost of shipping to Austria and will reply back to you later today.

Cheers Bill

ice-tii, 2022-08-10:
Hello Bill!

 

ok so i have the peorson to thank for all that good work, with Github!

I can speak for People around the Globe, thank you! Without your work it would be way more difficult to come into the World of printing and cnc.

 

So enought of that honey talk :-)

 

Time to get serios!

I can ashure you that it wasnt a grounding Problem. I have grounded every singele wire all on 1 ground just like a Technischian would do.

There where a lot of trouble because of me not knowing to ground wires, but my Father was there and showed me how to, hes IT spezialist!

 

So after i have grounded every singel Motor wire, the maschine and the Spindel, the only thing was still the PWM controll.

So i put a pwm converter outside to work like that and it worked just like with my v3, perfectly.

Then i shut of the spindel and pulled the plug from the v5 and messured the pwm and it was still going up and down.

My Dad said, that he likes the v5 but said thaqt some parts like the pwm converter sometimes might intervire with the rest of the electronics, so 

putting everything in one box can cause trouble. I even had troubel with the steps when i put 1/16 microstepping, just the z axis worked, but not Y and X.

I crashed my maschine to find that out :-)

 

I now have ordered the black box which i wanted to get in first place, becuase now i have to send the v5 back to Spain, have to pay the shipping and some reduce of the price,

because of me using it. So for myself, i dont want to put any more money on that, because if i will get a nother one which does the same i can go trough that process again

and lose time and money. And Shipping cost are exploding these days! :-)

 

So with the V3 you have, if you dont really need them and you want to get that box out of your way, i would take the box if the shipping is not to high.

The only thing for me is that the v3 is rated to 2,5 A for the steppermotors, which can cause problem, with loosing stepps will a job, especially, when you go to 1/16 microstepping.

But for beginners, thazt could be a good start, because i worked with the V3 now 2,5 Years and am doing amasing stuff. 

Mit freundlichen grüßen

Timo Helder
3D WELTEN
Ö/Vorarlberg
6943 Riefensberg
Riebinger 136

+43 6763632706
www.3d-welten.at

ATU74032806

 
 

Gesendet: Dienstag, 09. August 2022 um 17:15 Uhr

Spark-Concepts, 2022-08-13:
Timo,

Would you mind emailing me at bill@spark-concepts.com