Spark-Concepts/xPro-V5 #159: macro buttons

Bentowle, 2022-07-11:
im kinda confused how to setup the macros the way i want to. 

i want 1 as start/resume and the other as pause/hold .     what are the line of code do i enter todo this?

will the start button start the program after i load  the file? like a cycle start does on a haas?      would pause hold work the same as a haas ie i can press the feed hold and the program pause , can blow chips out etc then hit the cycle start/resume again?

Bentowle, 2022-07-11:
i understand $User/Macro0= and $User/Macro1=   are the two codes i need to enter.   i dont know what they should equal or if i need to equal a few things.

Spark-Concepts, 2022-08-08:
I'm assuming this has been resolved - if not, please re-open this issue...

I believe the latest firmware revision has this defined already [CNC_xPRO_V5_XYYZ_NO.bin](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NO.bin)

Otherwise you'll need to do the following:

This setting be made in the  in the ESP32Grbl [source code](https://github.com/Spark-Concepts/xPro-V5-Firmware/tree/main/CNCxPROv5_20220808/src) prior to compilation - the firmware would be done when compiled: [_How to Compile via Arduino_](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-Arduino-IDE)

Assuming the default machine config (normal XYYZ with a normally open e-stop switch) : 
Changes made to [CNC_xPRO_V5_XYYZ_NO.h](https://github.com/Spark-Concepts/xPro-V5-Firmware/blob/main/CNCxPROv5_20220808/src/Machines/CNC_xPRO_V5_XYYZ_NO.h)
```cpp
/***********   Input Pins  ******************/
#define PROBE_PIN               GPIO_NUM_22
#define X_LIMIT_PIN             GPIO_NUM_35
#define Y2_LIMIT_PIN            GPIO_NUM_36
#define Y_LIMIT_PIN             GPIO_NUM_34
#define Z_LIMIT_PIN             GPIO_NUM_39

// Comment out the two lines below 
// #define MACRO_BUTTON_1_PIN      GPIO_NUM_13
// #define MACRO_BUTTON_2_PIN      GPIO_NUM_0   

// Then add the following two lines   
#define CONTROL_FEED_HOLD_PIN      GPIO_NUM_13
#define CONTROL_CYCLE_START_PIN      GPIO_NUM_0 

#define CONTROL_SAFETY_DOOR_PIN GPIO_NUM_16
/********************************************/
```

Make sure to define the machine type on the [Machine.h](https://github.com/Spark-Concepts/xPro-V5-Firmware/blob/main/CNCxPROv5_20220808/src/Machine.h) file:

```cpp
// Find and Comment out the line below
//  #    include "Machines/CNC_xPRO_V5_XYYZ_NC_Y2Lim.h"

// Add the line below
#    include "Machines/[CNC_xPRO_V5_XYYZ_NO.h"
```

_note: change "CNC_xPRO_V5_XYYZ_NO.h" to whatever your machine type is if it's not the standard default load... i.e. CNC_xPRO_V5_XYYZ_NC.h_

closed 2022-08-08 by Spark-Concepts