Spark-Concepts/xPro-V5 #160: problemas para activar puerto y2 o que funcione.

jose1094, 2022-07-11:
hola espero estén bien necesito una ayuda tengo mi controlador xpro-5 pero el puerto y2 no esta activo o no manda señal al motor, no tiene ningún mensaje de error y si cambio el cableado de puerto a y el motor funciona perfectamente. espero me puedan ayudar 
( tenia otro controlador y se daño y lo cambiaron por este y apenas lo estoy configurando )
muchas gracias.

Spark-Concepts, 2022-08-09:
Please try updating to the most current firmware version as of July 2022.  If you are still having issues, please contact us via the "contact us" link on our homepage at spark-concepts.com

closed 2022-08-09 by Spark-Concepts

mattydboom, 2022-08-09:
> Please try updating to the most current firmware version as of July 2022. If you are still having issues, please contact us via the "contact us" link on our homepage at spark-concepts.com

Just to check as you mentioned here that latest firmware version is July 2022... but from the firmware page I can only find ones posted as recent as 27th June 2022. Is this one of the same or am I looking in the wrong place? Thanks