Spark-Concepts/xPro-V5 #163: Hello,,im using xprov5

Jennywhite2, 2022-07-16:
[image]
Did my wiring is correct,,im put Ac power to power supply then loop/jump to vfd,, im using xpro v5,,,can some one shari g worong diagram xprov5,,,because on githut im not clear about wiring..

RoboDaRock, 2022-07-30:
The photos are not clear enough to make out. Can you take some close up shots.

spydercnc, 2022-07-30:
The other issue I had originally is that I used 4 core wire. Using a twisted pair, shielded, wire sorted that out as well. I have also put a 100ohm resistor over the terminals, although it was working without it.

RoboDaRock, 2022-07-31:
Hi spydercnc can you email me the firmware file you used my email roccoszabo@gmail.com thanks in advance

RoboDaRock, 2022-07-31:
Hi spydercnc  any chance you could email me the firmware you use  thanks in
advance

Spark-Concepts, 2022-08-09:
RoboDaRock,

Were you able to sort things out? If not please post a more clear image of your current wiring - also please review 
 https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1208726453

Spark-Concepts, 2022-08-09:
RoboDaRock,

That was my mistake, that last comment requesting images was directed toward Jennywhite2. I see you re-posted on the other issues thread. I totally get and don't blame you for being frustrated.

I believe we are shipping you a new unit which I personally tested on our rig via multiple 3hr test jobs running a rs485 spindle while monitoring and verifying all system functions before packaging and shipping.

closed 2022-11-09 by Spark-Concepts