Spark-Concepts/xPro-V5 #164: Problem connecting VFD spindle 2.2 kw water cooled to xprov5

Nmfinest, 2022-07-28:
Hello everyone I'm having a problem getting my cncj software to recognize or to get my spindle to turn on I've tried to update firmware reset VFD resetting my xprov5 I'm at a lost and have been using my spindle manually which isn't bad just would like it run right so if any one has any suggestions please let me know thank you

RoboDaRock, 2022-07-30:
Hi yes I'm having the same problem. It worked for me for about 2 hours then nothing. First of all check Wether the light under the pwm/rs485 switch is flashing green as well as red. This way you know that they are both talking to each other . Set up the vfd prg 1 is set to 2 and prg 2 is set to 2.  Foll ow the setup guide for other settings for the vfd.    If this does not work and you can spend alot of time trying to fix this issue like me. I downloaded a small program called spindle talker.  I bought a cheap usb to rs485 converter . Downloaded and installed the driver for spindle talker . Once I did this the program spindle talker controlled the spindle no problems. So this only leaves one thing. Since the vfd is working properly over rs485 through spindle talker that only leaves a faulty xpro. I've literally wasted months trying to source the problem. At least now I know the xpro is not working.  I'm just not sure a new one will fix it.

spydercnc, 2022-07-30:
I am using cnc3D commander. Used that to reinstall the original firmware to xPro and it worked, after stopping with the new firmware.

Craig Mclaren

RoboDaRock, 2022-07-30:
Yes I checked out cnc3d.  Wow leaves cncjs for dead.  I'll give the old firmware a go tomorrow let you know how I go. Thanks for the tip

Nmfinest, 2022-07-30:
Where did you get the USB to rs485

RoboDaRock, 2022-07-31:
Ebay

RoboDaRock, 2022-07-31:
Can you give me a link to the firmware you used. I tried changing the firmware with no luck. Used cnc3d commander no problems. Beats cncjs by a country mile.    Thanks in advance

Nmfinest, 2022-07-31:
How do u connect cnc3d to xprov5

RoboDaRock, 2022-07-31:
It uses the same drivers as cncjs and just press connect button

Nmfinest, 2022-08-02:
How do u connect cnc3d to xprov5 same way as cncj and then how did you get
you spindle working once you connected to the cnc3d

RoboDaRock, 2022-08-02:
I still haven't got the spindle working. To connect to cnc3d is the same as
cncjs only a bit easier.  I can only get the spindle working through
spindle talker. Sparks is sending me a new controller  I'm suspecting the
controller is faulty

Spark-Concepts, 2022-08-08:
All, 

Please read below in it's entirety (also for 220V VFD user please see note at the end of this post) - also, if at any point I'm not clear or what I'm saying doesn't make sense, please repost and I will do my best to explain in more detail.   

we're seeing a-lot of issues regarding RS485 communication errors.... most all issues regarding this revolve around folks with 220VAC mains and 90% of the issues are related to ground loops, RF coupling and/or faulty wiring or misplaced grounds/neutrals. These can sometimes be extremely frustrating and often difficult to troubleshoot . Some folks have also used RS485 to USB converters to test communications between the VFD without issue - in this situation your computer's power supply is typically isolated from mains power; in other words  you eliminate the potential for ground loop disturbances on the RS485 signal path.

RS485 is a differential pair, meaning the differential voltage between RS485A and RS458B _and not with referenced to ground_); this means common mode noise (noise the the same on both A and B and can dance around in sync if measured with an oscilloscope relative to ground - however if measured differentially (voltage on A - B) you should get a nice clean square wave... again this all falls short when coupled to ground.

 - RS485 is also bi-directional, meaning it sends and receives on the same set of wires

A few things to check first:
- ensure your [firmware](https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware) is up-to-date
- RS485 wiring is less than 1 meter in length, twisted, and connected properly (resistors are optional)
[image]
- Verify the **PWM-RS485** switch<sub>(1)</sub> is set to **RS485**
- Sometimes the PWM-RS485 switch contacts may be contaminated with flux as received from manufacturing; try using contact cleaner or applying 99% isopropyl alcohol with a dropper to the switch (working it back an forth a few times) _ONLY USE (99% IPA - ANYTHING LESS THAT IS DILUTED WITH WATER AND WILL HAVE THE ADVERSE EFFECT)_
- verify the spindle type is defined - on the command window send the following: ```$Spindle/Type=HUANYANG``` if that doesn't work, try ```$Spindle/Type=H2A```
- Verify your VFD is configured properly using the procedure in the [Huanyang-VFD-manual](https://github.com/Spark-Concepts/xPro-V5/files/6247012/Huanyang-VFD-manual.pdf) _PLEASE REFER TO THE MANUAL SPECIFIC TO YOUR MODEL VFD; not all VFD's are the same despite the name on it_

Set the following register values via the VFD touchpad (should be the same on most VFD's - VERIFY ON YOUR SPECIFIC MODEL VFD):
| Register  | Value | Description |
| :-------: | :-----: | ------------- |
 | PD000 | 0 | **unlock parameters** |
 | PD001 | 2 | **Command source is RS485** |
 | PD002 | 2 | **Speed source is RS485** |
 | PD163 | 1 | **Communications address 1** |
 | PD164 | 1 | **9600 b/s** |
 | PD165 | 3 | **8 Bit No Parity - RTU** |
 | PD000 | 1 | **lock parameters** |
 
 _Also check the following settings (these will setting will vary based on the values indicated on the motor's nameplate and your mains voltage 110VAC vs 220VAC)_
 
| Register  | Description |
| :-------: | ----- |
| PD004 | **Base frequency as listed on spindle** _(typically 400)_ |
| PD005 | **Maximum frequency Hz** _(typical value for spindles is 400)_ |
| PD011 | **Min speed** _(recommended air-cooled=120 water=100)_ |
| PD014 | **Acceleration time** _(test to optimize)_ |
| PD015 | **Deceleration time** _(test to optimize)_ |
| PD023 | **Reverse run enabled** _(set to 1)_ |
| PD141 | **Spindle max rated voltage** _(typically 220)_ |
| PD142 | **Max rated motor current** _(0.8kw=3.7, 1.5kw=7.0, 2.2kw=??)_ |
| PD143 | **Motor poles** _(typically 2 or 4)_ |
| PD144 | **Rated motor revolution at 50Hz** _(typically 3000 @ 50Hz and 24000 @ 400Hz)_ |
 
_as a side note, we (Spark Concepts) seldom run into issues with VFD's running on 110V mains.  Since we are US based (110V mains), it's impossible to test the 220V 100% representative. Therefore we are in the process of acquiring a 220V voltage converter and 220V VFD since the 220V model VFD is different from the 120V version_

 - Our to-do plan: reverse engineer the RS485 signal path on the most popular 220V model VFD's and will determine if/where coupling may be occurring and  try to reproduce/isolate any potential issues, and provide potential solutions.

RoboDaRock, 2022-08-09:
I haven't been able to get this thing working properly I've followed installation instructions to a T.  I've got 2 ferrite chokes on the wires as  shown. I've wasted months on this thing.  I'm totally fed up with trying to get  this thing to work properly.  
[image]
[image]
[image]

id192313, 2022-08-09:
I have the same issue but managed to get it to work after implementing all the fixes that have suggested before. In the end the only thing that make it works is the resistor on the RS485 (both side) and using a Belkin surge protector. The moment I use a different extension without any grounding it failed. Hope this help. 
[image]

mattydboom, 2022-08-09:
Is there any reason not to use a **Metal film resistor** rather than a Carbon film type (which looks to be shown in the diagram)?

I can easily get hold of the Metal Film type in both 100 & 120 ohm... so am looking to get confirmation on the type and which of these two ratings would be optimal before ordering. Thanks in advance for anyone's better knowledge on this!

Spark-Concepts, 2022-08-09:
Either metal film or carbon film is perfectly fine.

Nmfinest, 2022-08-09:
You and me both I have mine set up just like yours and I have had this
issue for months also don't know what to do I hit up a guy from sparks and
he said he would help me out on this issue but hasn't responded yet but
when he does and we get this taken care of I'll pass the info on to you and
you vise versa thanks

Spark-Concepts, 2022-08-09:
Nmfinest, did you use the contact us link on the Spark-Concepts.com site? If not, please use the contact us method to email us. Unfortunately if we were to throw our emails out to the entire internet, we'd be bombarded with spam email. Sometimes customers personal email's will get marked as spam and you think we're ignoring you. I promise that's not the case. 

If you sent us a message and it's been more than 2 business days and one of us hasn't responded, please use the contact us to poke us for status (sometimes personal emails get marked as spam and we miss them). In the meantime will talk with mike to see if he's aware of your situation.

Spark-Concepts, 2022-08-18:
If you've tried everything and are still having EMI issues (unreliable communication between the xPro and your VFD).  You may try installing an isolated RS-485 repeater **(with this device RS485 communication is completely isolated and provides additional ESD protection)**

~$15US via ebay/Amazon
[image]
[image]

RoboDaRock, 2022-08-18:
That is ironic that you should suggest this . I hooked up the xpro to the
rs485 adapter and it sprung into action. Red and Green lights going off.
This clearly shows the xpro is working fine. It wouldn't talk to the
Spindle talker program because it's configured as a master not a slave .
The common denominator in this is the isolation of the laptop I'm using to
drive the rs485 usb adapter.  I'm not going to get too excited yet. I've
ordered special rs485 cable from China and when it arrives I'll rig it up
and see how it goes. The main thing out of this is Sparks being proactive
to find a solution.

Cheers

Spark-Concepts, 2022-08-25:
1. Placing an Oscilloscope at the RS485 output signal from the xPro ```DISCONNECTED from VFD```:
    [image]

2. Measuring RS485 signal at the 220V VFD while connected to the xPro ```WITHOUT RS485 isolation```:
    [image]

3. Measuring RS485 signal at the VFD while connected to the xPro ```WITH RS485 isolation and twisted pair```:
   [image]
   [image]
   [image]

VossAero, 2022-08-25:
Is the second example with or w/o twisted pair? The twist is doing a lot.
What about resistors? not used at all?
Shielding? Not used?
What is the difference between the three graphs? What are they?

Spark-Concepts, 2022-08-26:
The second example is not using a twisted pair or resistors. You are accurate, the twist does improve the signal quality, but not nearly the amount the isolator does (difference is illustrated between image 2 and image 3 above). Also note, even with the degraded signal quality (image 2),  I was still able to communicate with the spindle. 

***NOTE: The resistors were minimally effective  in reducing noise with cable lengths less than 1-meter long***

Ultimately, the Huanyang VFD's due to the nature of how they are constructed, share a common ground and can unintentionally create ground loops thus permitting radiated noise to degrade the rs485 signal despite all best efforts.
[image]
Ultimately to mitigate this scenario I reccomend implementing an RS485 isolator as shown above.

I will post a much more extensive write-up this weekend illustrating the effects of each wiring scheme along with actual pictures showing the configuration of each.

VossAero, 2022-09-13:
I have ordered and installed the RS485 isolator/repeater. Unfortunately it does not help me.
In the beginning it works fine - leds on XproV5 and repeater show frequent communication. As soon as I start spindle rotation the communication breaks down >> communication leds blinking frequency goes down. No response messages in CNCjs until Alarm message. Never reacts to a M5 command. That all stops when I turn off the spindle by the STOP button on the VDF front panel.

I have the RS485 wire a twisted pair, I have resisitors, I have a 220V home supply, I have the spindle cable shielded and shield grounded to the VDF. I have VDF settings as instructed. No more ideas...

RoboDaRock, 2022-09-14:
I finally got my rs485 isolator wired it up. Reconfigured everything And
away it went. I'm hoping that this is a permanent fix.  Thanks for the help
Mike and the crew at sparks. Cheers

leadsmkt, 2022-09-15:
For me, same config  and same problem than @VossAero  with this RS485 repeater : https://amzn.eu/d/32Xcsbg

VossAero, 2022-09-15:
> For me, same config and same problem than @VossAero with this RS485 repeater : https://amzn.eu/d/32Xcsbg

What you bought is the wrong product. It is a repeater only but not an isolator - the isolator function is the important one I understand.

VossAero, 2022-10-07:
I have my solution! 
I was really trying everything. Including ferrits and so on. Also just putting the RS485 repeater/isolater did not bring any change.
I still was also using different wires and when I tried the last wire I had for test - surprisingly it started to work correctly. It is a audio cable with two drolled lines and a metallic shielding. The wire gauge is fairly small.
I removed all things that should help to solve my issues and found out that finally this specific wire together with the repeater/isolator box is doing the job. Nothing else needed, nothing else helping.
Now I can concentrate on using my router... great, but EMI is difficult game.

VitaliiSadovshikov, 2023-01-23:
I have ordered and installed the RS485 isolator/repeater (in accordance with the scheme attached in this thread), but the controller doesn't show communication (the controller led flashes with red light). When I connect the VFD and controller simply just with a cable the connection works. Does someone have the proble with connection of the RS485 isolator/repeater?

cdo1955, 2023-03-08:
I have the same set up with the RS 485 isolator. It was still sporadic until I put the spindle controller in a electrical box and grounded the box. The box has a slot on the bottom I left open for air flow. Ever since everything is working fine. My wave forms look much cleaner now that the source of the EMF in in a metal box. box type listed below.
[image]