Spark-Concepts/xPro-V5 #165: FluidNC

lagnat, 2022-07-29:
I saw in an older post that you're moving over to FluidNC.  I've been using v3.4.4 on the XPro for a while but newer versions do not work.  Do you have any update for FluidNC support?  The maintainers of FluidNC have stated that Spark-Concepts offered to send them a unit but nothing ever came of that.  They state that it's the only way they can fix the problem.  Can you arrange to send them a V5?  FluidNC is really very good and official support would be ideal.

mathojojo, 2022-08-03:
Hello,

Would be great yes. Especially because FluiNC let you easily change your machine setup if needed, without recompilation. It would clearly make the xPro better, and customizable for everyone.

spydercnc, 2022-08-05:
Cnc3D commander does that with a few clicks as well. They are doing a lot of work to make the xPro more user friendly…yet not supported by sparks?

Craig Mclaren

mathojojo, 2022-08-06:
Hello,

xPro V5 is reported as totally supported by CNC3D Commander

[https://www.cnc3d.com.au/commander](https://www.cnc3d.com.au/commander)

lagnat, 2022-08-07:
@Spark-Concepts I don't think it's asking too much for some kind of status on this given it's something you have said you plan to support.

Spark-Concepts, 2022-08-07:
FluidNC is a vastly different construct (code-wise) and massive effort for the developers. We support the effort. Up to now Spark Concepts has not made any claims as to moving over to FluidNC near term. That said, there may be other folks leaving speculative comments mentioning otherwise. 

Update:
We have sent two units over to the main FluidNC developers. I've confirmed today that at least one as been recieved and they are in the process of evaluating for compatability. As of now I can not provide a definitive timeline as to when that'll be completed. However, we are working closely with them to provide them whatever they need moving forward. 

As for now I would patiently keep an eye on the [FluidNC github site](https://github.com/bdring/FluidNC) for the most up to date status. I would also highly encourage you to [support](https://www.paypal.com/donate/?hosted_button_id=8DYLB6ZYYDG7Y) their hard work.  These guys are awesome!

_note: any beta-release claiming compatibility will be posted via the FluidNC github site... with that we can make no claims about FluidNC until we have extensively tested it ourselves and have fully vetted it_

lagnat, 2022-08-07:
Well that's really great news!  Bart and Mitch have done some really incredible work with FluidNC.

mathojojo, 2022-08-07:
Great news yes, as long as it is working at least as good as actually, maybe even better.

ImagineerNL, 2022-08-07:
@Spark-Concepts may i suggest defining a statement regarding support of FluidNC and posting that  on the primary page of this repo?

I have been messing and searching to what it was i am doing wrong in being unable to update to the newest firmware and finally stumbled upon this issue.

The current link on the frontpage of the webui of your own xPro V5 box (freshly unpacked and working on my install) links to the [ESP32 repository page](https://github.com/bdring/Grbl_Esp32)
[image]

and that repository page has a big main header 
> The next generation of Grbl_ESP32 was such a massive upgrade we decided to change the name. It is called FluidNC and is [available here](https://github.com/bdring/FluidNC). Please check it out and give us a star. It is compatible with all Grbl_ESP32 hardware.

Therefor I, and probably others as well, are automatically directed onto a path that is currently directly opposite of what you are stating in the text above. 

Also, it is worth mentioning that the 'future' of the xPro will be depending on your support of FluidNC, since the xPro V5 is built on Grbl_Esp32 and that is stated by its developers as 'This version is only being maintained with existing features. All new features are targeted at FluidNC.'

That being said; great to hear you are currently in active contact with them and shipped devices out to them!
This is exactly why i chose the xPro over other non-esp32 boards; the potential is huge.

Slixxor, 2022-08-21:
Thanks for the feedback and support @spydercnc  :)

I have indeed focused very heavily on simplifying the relationship between the xPRO V5 and the user interface in Commander.

There are easy breakouts for all the extended features (SD card management, setting motor current, easily setting up wireless connections) and pretty much every $ setting in the GRBL setting space.

@mathojojo 
At this stage, Commander does not support FluidNC, the developers have removed almost all of the $ settings needed to populate the easy breakouts in Commander. I may consider changing this in the next release as a suitable, user friendly, sender to my knowledge does not exist right now.

@Spark-Concepts  does not recommend Commander for some reason despite it being likely the best possible Sender for the xPRO V5. It also has a lot of "user-proofing" alerts for new users which minimizes typical errors we've seen with new machine owners.

Maybe it's time your team seriously considered recommending Commander, It's totally free and is a good experience for your users and in turn will be a better experience for you too.

I'll keep working on that part anyway. :)

The new release for Commander is due very soon.  This time around I've added some CAM tools for pen plot and vinyl cutter and a few other goodies like a fully customizable tool management system. There is also easy access to our CNC3D ESP32 tool in the Help menu that can be used to upload firmware files via USB. We've acknowledged that sometimes firmware over SD or via upload fails, this will work regardless of these issues.

We're aiming to launch the new release this week with the new features but first we need to finalize our own Auto firmware update for our own controller. This the first time we're updating our firmware and so far everything is going to plan. :D

If all goes well this launch; @Spark-Concepts is welcome to email me directly to implement any of their firmware enhancements in a BIN file and that way users with xPROs can easily get the latest firmware without too much fuss. They can also use the CNC3D ESP32 tool as a backup if something goes wrong.

The auto firmware update tool works on both USB and Wifi.