Spark-Concepts/xPro-V5 #166: Update to WebUI V3 ?

mathojojo, 2022-08-03:
Hello,

Would it be possible to upgrade firmwares with the last available version of WebUI (v3). It seems to be a nice upgrade, with more options, better integration, and can fit perfectly Smartphones and Tablet design.

[https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Features.md](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Features.md)
[image]

Thank you

Spark-Concepts, 2022-08-07:
Mathojojo,

I will definitely revisit WebUI (v3) - we've been aware of it for a while; development on the GRBL adaptation has been very slow going.  Once it's been complete - I agree the integration will be very nice.

If you haven't updated the webUI from the stock load - there is an up-to-date and much cleaner version for Smartphone and Tablet available here: [index.html.gz](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/index.html.gz)

## OTA (Over The Air) WebUI Update:

To upload/update the WebUI, upload the  [index.html.gz](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/index.html.gz) via WiFi. With the WebUI open, click on the ESP3D tab. Now click on the yellow cloud icon. 

[image]

- Delete old index.html.gz and select and upload new [index.html.gz](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/index.html.gz)

[image]

- Once the upload is complete, you'll need to reset and reconnect to the xPro to view the new WebUI

_NOTE: these are files that are stored on the ESP32 SPIFFS (SPI Flash File System).  These files are part of the ESP3D WEBUI_

mathojojo, 2022-08-07:
Again a good news for the WebUI v3, thank you.

For the last index.html.gz, thank you as well. For the replacement process, I just correct that the green button should be clicked, not the yellow (Just to be sure that someone doesn't dtry to flash the firmware with an html.
[image]

Ugilten, 2022-08-16:
Hi, I tried updating the index.html.gz file like described above. But with my 10 thumbs I uploaded the wrong file and now I don’t know how to get the webUI back.
I tried pulling the SD card out and replacing the index file on the computer, but that didn’t make any difference.
Is there any way to restore the webUI? 
Regards

Spark-Concepts, 2022-08-17:
With our xpro powered and your computer connected via wifi, try to open the following URL in your web browser. 
http://192.168.1.69/?forcefallback=yes (use the address of your ESP32)

- next, you should be prompted to load the data file that contains the WebUI
- Use the Choose Files button to select that file, then click the Upload button
- Once the file has loaded, refresh your browser

Ugilten, 2022-08-17:
Hi @Spark-Concepts, That did the trick. I am now up and running again, Thank you.