Spark-Concepts/xPro-V5 #167: Are sources up to date for compiling ?

mathojojo, 2022-08-07:
Hello,

I see that last sources modifications for the xPro V5 (here : [https://github.com/Spark-Concepts/xPro-V5-Firmware](https://github.com/Spark-Concepts/xPro-V5-Firmware)) were modified on 2021 November.

Your precompiled firmwares had 2 updates April and June :

April release Note :
- Minor bug fixes
- TMC5160 Tuning
- Fixes bug requiring double reset for Z motor

June release Note :
- minor bug fixes and motor/current tuning

I successfully compiled the sources, with your CNC_xPRO_V5_Machine_Template.h file without changing anything, just to see if I was able to do that (I never compiled anything up to now) : successfully, a good point.
My goal is to adapt it to my motors (run an hold currents, in a first step).

But : 
Bugs (and optimizations) were fixed in April and June 2022, but I compiled from sources from 2021 november.

Do my self-compiled binary still suffer from the bugs described above, or not ?? (no TMC5160 tuning, no fix for the double Z motor reset, no motor current tuning, and no optimizations )
 If yes, you should really considere to update your sources, because they become unusable and useless if they are blocked in a buggy state.

Thank you for your consideration and help.

Spark-Concepts, 2022-08-07:
mathjojo, good catch & thank you for the heads-up.

You are correct, the source files do not match the most up to date compiled bin files - I will post the most current source code here very shortly (I'm not at my computer at the moment).  I will re-comment on this issue once I've posted the update source code

closed 2022-08-07 by Spark-Concepts

mathojojo, 2022-08-07:
Thank you very much, it's greatly appreciated.

Maybe you can in the same time answer a question: 
On the [GRBL_ESP32](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-Arduino-IDE#compiling-firmware) documentation, they say we should use the v1.03 version, because the 1.04 and above have a bug. But many versions have been published, and now we have the 2.04 available.

Do you still use 1.03 to compile your precompiled firmwares, or is it now possible with last versions ?

Thank you again.

Spark-Concepts, 2022-08-08:
mathojojo,

The most up-to date source code files have been posted on the [https://github.com/Spark-Concepts/xPro-V5-Firmware](https://github.com/Spark-Concepts/xPro-V5-Firmware). Download the files found [here](https://github.com/Spark-Concepts/xPro-V5-Firmware/tree/main/CNCxPROv5_20220808) into a folder and add as a library to Arduino.

To answer your question use esp32 board libraries version is **v1.03** - as of May 2022 the latest release is still buggy.  My co-engineer at Spark (Mike) prefers to use the Arduino IDE - whereas I prefer PlaformIO & VS Code ([PlatformIO](https://platformio.org/) microcontroller compilation system, coupled with the [Visual Studio Code](https://code.visualstudio.com/) (VSCode) editing system.

That said, use the one you're more familiar with... 

Bart has some good resources over on his github page regarding compiling via Arduino-[(here)](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-Arduino-IDE) and PlatformIO/VS Code-[(here)](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-PlatformIO)

mathojojo, 2022-08-08:
Thank you for everything.